prusa3d/Prusa-Firmware #1461: M106 command blips the fan to 100% temporarily

zsoltmar, 2019-01-17:
Whenever in an off state (M107), any value M106 Sxxx will first ramp up the fan speed to 100% then lower it to the desired speed. How to disable this? Where and when did this behavior get introduced?
This behavior makes PWM applications unusable. Any insight would be highly appreciated!

AnHardt, 2019-01-17:
9 Jan 2013
```
// When first starting the main fan, run it at full speed for the
// given number of milliseconds.  This gets the fan spinning reliably
// before setting a PWM value. (Does not work with software PWM for fan on Sanguinololu)
//#define FAN_KICKSTART_TIME 100
// When first starting the main fan, run it at full speed for the
// given number of milliseconds.  This gets the fan spinning reliably
// before setting a PWM value. (Does not work with software PWM for fan on Sanguinololu)
//#define FAN_KICKSTART_TIME 100
```

AnHardt, 2019-01-17:
Sorry wrong repository.

zsoltmar, 2019-01-17:
Much appreciated!