prusa3d/Prusa-Firmware #1471: Feature Request: Message parameter for M600

rainersc, 2019-01-22:
I created myself a Printer Profile that has 2 or more Extruders and uses a manual tool change GCode to change colors using M600. This allows multi color prints from Slic3r with manual M600 based filament changes (without having a MMU)
For 2 colors this works fine, but if i want to use more than 2 colors, the user would not know to which extruder/filament a change is requested.
A simple possibility to pass an additional Message like "change  from [previous_extruder] to [next_extruder]"
would be able to resolve this.
Any chance to get this added?

workinghard, 2019-01-22:
Please check if placing M117 "message" before M600 will work for you.

rainersc, 2019-01-23:
i tried m117 before posting here, but m117 does only display a message, does not wait for user input and the message is not visible in the moment where m117 is being called. so M117 does not help.
i checked as well: M1, m601/602, but noe provided a working solution

workinghard, 2019-01-23:
```
M117 "message"
M0
M600
```
Should display the message and wait for button press. Once button is pressed, it will initiate M600. But I think this is not what you want because printer will sit on a perimeter and wait.

rainersc, 2019-01-23:
yes. there are 2 problems with that apoarch.
1. printer does not go to the "filament change position"
2. M0/M1 are not working good with octoprint 
i still think that when calling M600 to ask for a filament change it is obvious that there should be a possibility to tell the user "which" change is required.
passing a new parameter and "S change from to [next_extruder]" would give a clear indication of what is expected from the user and would be at the correct part of the workflow.

rainersc, 2019-01-24:
one alternative way to resolve it would be if M600 would not clear out the message from a previous M117 command

cool-RR, 2019-01-26:
Funny, I just had the idea for this feature while doing a 6-color print. I started writing a feature request, and saw that this one exists and was opened just 4 days ago. So count me as a +1 to this feature request.

rainersc, 2019-01-27:
just to explain a bit better why this would be good: i created a tool change gcode which adds these lines into the gcode for the different tool change positions:
M117 change to 1 red
M117 change to 2 black
M117 change to 0 yellow
M117 change to 1 red
M117 change to 0 yellow

but the message only gets displayed for 2-3 seconds while the extruder moves to the m600 filament change position.
If it would be still visible at the beginning of m600 this would help a lot and add more value to the printer for me..

rainersc, 2019-03-05:
Here is the multicolor profile i am working on, which might explain why it would make sense to add this feature..
[https://schlosshan.eu/blog/2019/03/02/prusa-i3-mk3-real-multicolour-prints-without-mmu/]

rainersc, 2019-03-07:
Thanks to the work of guy.k2, we have a working prototype with a slighltly different aproach.
He suggested a "M600 Q" parameter, which when give removes the initial
M600_wait_for_user does
lcd_display_message_fullscreen_P(_T(MSG_PRESS_TO_UNLOAD));
message.
This allows us to call something like
```
M117 Change to Yellow
M600 Q
```

There are 2 compiled test versions available ( for standard MK3 and for Bondtech variant).
more details can be found in this thread: https://shop.prusa3d.com/forum/user-mods-octoprint-enclosures-nozzles--f65/simple-way-to-print-in-real-multi-color-without-mm-t28186.html

guykuo, 2019-03-07:
I'm a total newbie on github and have not learned how to properly branch and send pull requests. It's a pretty trivial bit of coding needed to implement the  M600 Q parameter.  Here are the relevant changes. I have commented the new/modified lines and included enough code lines around those to find the correct location in the existing files.

I would try to do a branch and submit myself, but I don't know github well enough to do so.
Anyways, this is what we have working....

in Marlin.h

`
void M600_load_filament();
void M600_load_filament_movements();
void M600_wait_for_user(float HotendTempBckp, bool suppressInitialMessage); //Kuo <---for M600 Q
`

in Marlin_main.cpp

`
=== in the case 600 statement

    case 600: //Pause for filament change X[pos] Y[pos] Z[relative lift] E[initial retract] L[later retract distance for removal]

	float e_shift_late = 0;
	bool automatic = false;
    	bool suppressInitialMessage = false;  //Kuo <---for M600 Q

          #ifdef FILAMENTCHANGE_YPOS
            y_position = FILAMENTCHANGE_YPOS ;
          #endif
        }

    	if(code_seen('Q')) 				//Kuo <---for M600 Q
        	{ 					//Kuo <---for M600 Q
          		suppressInitialMessage = true; 	//Kuo <---for M600 Q
        	}					//Kuo <---for M600 Q
        
	if (mmu_enabled && code_seen("AUTO"))
			automatic = true;

	gcode_M600(automatic, x_position, y_position, z_shift, e_shift_init, e_shift_late, suppressInitialMessage);  //Kuo <---for M600 Q

=== in static void gcode_M600

static void gcode_M600(bool automatic, float x_position, float y_position, float z_shift, float e_shift, float /*e_shift_late*/, bool suppressInitialMessage) //Kuo <---for M600 Q

    //Beep, manage nozzle heater and wait for user to start unload filament
    if(!mmu_enabled) M600_wait_for_user(HotendTempBckp, suppressInitialMessage); //Kuo <---for M600 Q

=== in void gcode_M600

void M600_wait_for_user(float HotendTempBckp, bool suppressInitialMessage) { //Kuo <---for M600 Q
    //kuo added suppressInitialMessage
		KEEPALIVE_STATE(PAUSED_FOR_USER);

		int counterBeep = 0;
		unsigned long waiting_start_time = millis();
		uint8_t wait_for_user_state = 0;
    
		if (!suppressInitialMessage) lcd_display_message_fullscreen_P(_T(MSG_PRESS_TO_UNLOAD));  //Kuo <---for M600 Q

`

rainersc, 2019-03-07:
i already tested the above change in a custom firmware that guykuo compiled for me, and i can confirm it works as expected and would be a perfect working solution for this feature request
@inquisitiveKangaroo can you perhaps help with getting this progressed?

guykuo, 2019-03-08:
I learned how to create a branch and submit a request. However, I can do so **only** on my fork of the MK3 firmware. Does one need to be invited to gain the ability to create new branches for submittal to the  MK3 repository?

stahlfabrik, 2019-03-08:
Yes of course. Please Read about Pull Requests

guykuo, 2019-03-08:
Yes, I understand how to **create a pull request**, but I can only create pull requests from branches off my fork back to my fork. That keeps the MK3 repository safe.

I wish to create a branch that can be used to submit a pull request back the main repository. So I can help with this issue.

Who do I ask for permission to create a branch off the main MK3 repository? 

Alternatively, is there (not so obvious) method to submit a pull request to the main MK3 repository from a branch of my fork?

stahlfabrik, 2019-03-08:
Yes, that is how pull requests work. You fork, you push to your fork. Then create a pull request

guykuo, 2019-03-08:
Yes I more than get that.

My quandary is not about how to create a pull request, but how to make the pull request to someone _other than to myself in my own fork_. 

I can create a pull request without difficulty to MYSELF, but see no what to send a pull request back to the MK3 main repository. That, of course, does nothing for getting anything back to the main repository.

You seem to be indicating that there IS a what to create a pull request from ones private fork back to the main repository. That is what I'm somehow missing.

The alternative would be branching from the main repository, but I don't have permission to do that.

stahlfabrik, 2019-03-08:
You usually create a separate branch for a pull request that only has the changes you want to have pulled. Otherwise changes of getting it pulled are close to zero. :-) focus is very important 

And the number of branches in your repository might be unlimited.

guykuo, 2019-03-09:
Rainer, I hope you don't mind, but I went ahead and put in a pull request for the M600 Q parameter

Had to delete my existing fork with all its 0.9 stepper mods. Create a new branch off it to implement the M600 Q and then submitted a pull request from that branch.

I will recreate my 0.9 stepper work in a new branch of my repository.

rainersc, 2019-03-09:
perfect! thanks a lot for your work and help on this!

guykuo, 2019-08-31:
Prusa doesn't seem willing to look at this improvement, but I am arranging things so you can use with firmware 3.8.0

Old branch was deleted and new one created based on 3.8.0
https://github.com/guykuo/Prusa-Firmware/tree/M600-Q-enhancement

The new branch is mergeable with Prusa's MK3 branch

Even better you can also merge it into my 0.9 degree motor support branch at https://github.com/guykuo/Prusa-Firmware/tree/0.9-Degree-Stepper-Support

which adds a bunch of goodies of other extruders, hot ends, and 0.9 degree motors.

alexiri, 2021-06-05:
I just finished printing a 3-color piece and I must say it was way harder than it needed to be. I'm really looking forward to a solution to this issue that would at least show a message to help keep track of what filament to change to.

FrHePo, 2022-04-16:
Is there any news about adding this feature to a regular release?

gudnimg, 2022-04-16:
This reminds of an issue which is similar regarding the messages disappearing: https://github.com/prusa3d/Prusa-Firmware/issues/3316 in this case with M601

FrHePo, 2022-04-16:
Thanks for the reference!

Panayiotis-git, 2022-04-17:
A long time ago, I've created this https://github.com/prusa3d/Prusa-Firmware/pull/2380, that helped me a lot.

GithubUser99999999, 2022-04-30:
Is there any news on this feature getting implemented by Prusa into all of their printers? I am using a Mini+ and just using the original firmware would be my preference. 

Even Prusa Slicer still doesn't seem to contain this functionality without manually adding M600 code and then grabbing excess material with tweezers since it's also missing a stacked intro line setting for each filament change.

FrHePo, 2022-04-30:
People responsible for PrusaSlicer don't seem to be much interested in transparency concerning the software development :-((

closed 2023-09-14 by github-actions[bot]