prusa3d/Prusa-Firmware #1478: Issues during printing PolyMaker PC-MAX

landrunner22002, 2019-01-24:
Hello here a list of identified issue during operation: (firmwares 3.5.1/MMU 1.0.3) : Prusa MK2.5 + MMU2

Issue 1 : With "Polymaker PC-MAX" mode in Slic3r 1.41.2, printing start without problem with Bed temperature for the first layer at 100°C and switch to 115°C for the other layers. But temperature bed increase very slowly but reach 115°C despite that my prusa is in closed box with 40°C regulated temperature: printing fails with "Bed Temp Error" (temp stabilization problem) 
--> Is it possible to add a disable "Bed Temp Error" Time Out in menu (or increase timeout duration) ?

Issue 2 : Many times toggle button or/and display is not reactive : it's select another line after pushing due to very slow refresh listing on screen.
--> Is it possible to increase display refresh as like as when you turning toggle button ?

Issue 3 : After printer stop force action, when I select "unload filament" no message asking me to preheat before removing operation (stalling extruder motor).
--> Is it possible to remove filament into inital operation before stopping heating extruder ? 

Issue 4 : Bug identified when printer stop due to "bed temp error". After pushing reset "X" button, message is displayed from MMU2 "MMU needs user attention" and asking me to push button to resolve error but didn't tell us which button to push or following process to resolve the problem. Nothing happening after pushing each buttons of MMU2, so I have to cutoff printer, forcing to remove filament quickly before cooldown and switch on it to "resolve" issue...
--> Is it possible to have a detailed operation when "MMU needs user attention" is displayed ?

Thanks and regards
Stef

workinghard, 2019-01-25:
1. Should be resolved in the next release
2. Not sure if it's feasible. Sometimes you might also get a turn signal while clicking. 
3. As long as the nozzle temperature is high, Unloading is allowed and no error message appears. I think current limit is 180C. With PLA it works ok but you might get issues with PC-MAX with such a low temperature. IMHO: This is an operator issue, not firmware.
4. Maybe reset doesn't work properly in this case. It's allowed to have filament inserted during startup. So turning printer off and on should work in this case. Then do the preheat and unload filament.

landrunner22002, 2019-01-28:
Thank you for fast answering. About point 4: the extruder keeps its temperature operational despite "MMU needs user attention" message so I don't understand why I have to cut off printer instead off button action on MMU2 to resolve issue. 
It's also happening when you stop the printer by panel menu and relaunch it: printer initialization fails with leaving filament is noozle instead of unloading it...

landrunner22002, 2019-04-15:
Hello,

FW 3.7.0 on MK2.5 prusa : I've always bed temperature issue :( --> printing start without problem with Bed temperature for the first layer at 100°C and during switch to 115°C for the other layers: printing fails with "Bed Temp Error" (temp stabilization problem despite that my prusa is in closed box with 40°C)
--> No disable "Bed Temp Error" Time Out available in menu (or increase timeout duration between first layer and other layers)

closed 2023-09-14 by github-actions[bot]