prusa3d/Prusa-Firmware #1483: Printer says "recovering print" when it was not printing during power failure

xxv, 2019-01-26:
My printer seems to be stuck in a confused state where it is trying to recover from a print that never failed. When I power it on, it says, "recovering print" and then "Heating..." but the set temperatures are both 0. Pressing the reset button doesn't unstick this failed state.

I have my printer plugged into a z-wave controllable outlet switch and I turn it off to save power (it seems to draw 10W when idle). Last night I turned it off using the outlet switch and heard it beep loudly before it powered off; supposedly this was the event that triggered this state. I've powered it off with the outlet switch before and it hasn't done this prior.

The printer was not printing anything when the power was cut and my previous print had been done for a bit. I drive the printer via a raspberry pi through OctoPrint.

Firmware version: 3.5.1

xxv, 2019-01-26:
Unfortunately reflashing the firmware didn't wipe the necessary stuck bits and I still cannot use my printer. I'll be digging around to try to find something to clear the NV memory.

rambopierce, 2019-01-26:
I had a similar issue and the only way I found to clear it was to hold the knob down for a few seconds until it beeps which brings up the factory reset menu. I think I then selected reset all or some such. Pain because you have to go back through calibration.

crioux, 2019-02-09:
Also encountering this same issue. Had to factory reset.

ravishchawla, 2019-06-14:
Faced this issue twice in the last few days. It happens when a print fails because of a mechanical issue, and you shut down the power. The second time it failed, I had put the steel bed on incorrectly where it was slightly on top of the screws in the back, it caused a z-level failure (as it should), and then I took out the power. It's ridiculous because if a print fails you should turn off the printer, that's literally the only thing you can do, and if it's causing a bug, it's no good.

Thankfully I saved all of my live-z calibration squares this time so it shouldn't be too much of a hassle, but it sucks regardless.

phisher1, 2019-10-16:
This is happening to me as well.  3.8.0 MK3 -- Only on one out of two MK3. 
The printer gets caught in a "Recovering Print loop". It is mimicing a power loss, but there is none.  No hardware failure, no stuck fans. bed is fine. 
I did several 6+ hour prints in a row, then whammo, 2 hours in, "Recovering Print loop" 

I'm attempting a re-flash and factory reset.

FarbrorMartin, 2019-10-18:
I just encountered this problem. I turned off power at the end of a print to stop it from moving the printhead too high. This resulted in an endless Recovery loop with the printer just sitting doing nothing with the text "Heating..." I managed to get it out of it by raising bed temp to what I had used in that print, and inserting the sd card with the stopped print.

The obvious solution to this would be to just have an option of canceling Recovery mode. It could be useful even i real recovery situations,  because you might not want to keep printing. The way it works now is just dumb.

phisher1, 2019-10-19:
I downgraded to 3.7.0, fully power cycled. I've had numerous multi-hour successful prints since then.

pierrep, 2019-10-24:
I've just run into this problem after cancelling a print by switching the power off. Now it's stuck in recovery mode. Really dumb that all I can do now is a reset.

silicon-ghost, 2019-10-26:
I've been fighting this off and on since I upgraded to 3.8.0.  Repeating the same exact print will sometimes get me stuck in "recovering print".  I contacted Prusa support and the rep hadn't heard of the problem. 

I did watch the serial line to see the debug info during PrusaMK3S power cycles and it was "recovering" the print at the byte position just past the end of the print.  The remaining lines in the gcode file were all comments and blank lines put there by PrusaSlicer.

Reverting to 3.7.2 firmware didn't get me out of the "recovering print" boot loop and I was forced to factory reset.  I decided I could care less about power loss recovery and disconnected the power loss cable.  I'm still on 3.7.2 and happily printing.

FarbrorMartin, 2019-10-26:
Like I said before I managed to get out of the loop by heating the nozzle
and bed. I believe the loop was waiting for correct print temp, but it
didn't include the heat command, so it just waited forever.
I first tried to remove and insert the sd card, but that didn't work. Then
I set the temp to my print temp. That didn't work immediately, but when I
reinserted the sd card the printer got out of the loop and after a reset it
was back to normal.

So maybe this could be worked around by changing the gcode file that was
being printed. Find where it stopped and insert some command that does not
rely on anything else. Like G80 maybe.

pierrep, 2019-11-01:
@FarbrorMartin  thanks for the tip, heating the bed to the same temp as the "recovering" print did the trick.

Ninjarev, 2019-11-21:
OK guys i can confirm this is a bug never turn your printer off in middle of print if you dont want to get in this loop there is a way out of it without resetting everything just like pierrep said go to settings and set your print nozzle to 240c  after it reaches temp then turn off printer and back on and it will ask do you want to resume print click no and your back on track

rhtenhove, 2019-11-22:
What's even worse is that it even occurs after powering off after a successful print.

The printer sounds a loud beep when turning it off. When turning it back on, it displays the same behavior as described in previous comments, after inserting the SD card, or when still having the SD card inserted.

This happens on an MK3 upgraded to an MK3S, with firmware 3.8.1 (for the MK3S of course).

JT73, 2019-11-23:
Hey Prusa, first reporting in Jan, happening to a bunch of people. I just got my new printed and it's happening over and over to me also. Can I return this and get a refund please?

wavexx, 2020-01-27:
This hopefully should be fixed with #2437 which is new and just got merged, in combination with previous fixes (#2273, #2274) which are part of the 3.9.0 beta.

During automatic recovery (short power failure) it might look like you get stuck: "Stop print" only becomes available after the extruder starts moving again, which is not great (will be improved later hopefully). If you power off before that, the printer will keep trying to recover (it just assumes it lost power again).

AndersWinther-Dahl, 2020-02-08:
I did a first layer calibration (without adjusting anything) and the loop disappeared.

Herbiedc3, 2020-03-01:
Thankyou Anders.  Worked perfectly without the hassle of a reset.

closed 2023-09-14 by github-actions[bot]