prusa3d/Prusa-Firmware #1494: Stopping SD card prints sometimes leaves in bad state

mujido, 2019-01-31:
Printing from SD card and choosing stop print from the menu frequently leaves the printer in a strange state. Everything on the menu appears to work correctly. However if I go to print a model again the priming line printed at the bottom of build plate will be much longer than normal. Something like twice as long. This leads to other strange behaviors like dimensions seeming to be incorrect or Live Z adjust not adjusting to the correct level and/or not saving to permanently. These of course lead to failed prints. Doing a reset via front button will clear it from this state.

To get around this for the time being I am making sure to do a full reset every time I do stop print.

inquisitiveKangaroo, 2019-02-05:
Hi,
thank you for the report. Could you provide us gcode and the firmware version?
Kangaroo

jtnyansa, 2019-02-08:
Same issue also ends up causing the z leveling enforced issue

mujido, 2019-02-09:
Using firmware 3.5.0-1749 with the attached gcode.

[First_Layer_Square_0.2_0.2mm_PLA_MK3.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/2847520/First_Layer_Square_0.2_0.2mm_PLA_MK3.gcode.zip)

jventerprises, 2019-02-13:
same here, 3.5.0-1749.  i have noticed that when in this state the built-in first layer cal routine also misbehaves and instead of printing right angle lines forming rectangles, it prints triangles.   a reboot is required to fix.

wavexx, 2019-06-07:
I verified this also on 3.7.1. Stop a print and re-run the same gcode.

The priming line is printed incorrectly, and it's generally the first thing you notice: the double feedrate is changed into regular feedrate but with twice the length.

closed 2023-09-14 by github-actions[bot]