prusa3d/Prusa-Firmware #1512: Printing Stops Because of Arbitrary Character(s) without Newline

AustinSaintAubin, 2019-02-07:
Discovered a bug with the [Prusa Firmware v3.5.1-1778](http://prusa3dprinters.vshcdn.net/downloads/firmware/prusa3d_fw_3_5_1_MK3.zip)
Printer seems to halt if statement not ending in newline is received. 
Tested by connecting to [ESP3D](https://github.com/luc-github/ESP3D) and sending commands over telnet.

**Any one of the following commands will halt the printer.**
```
{ perl -e' print "\x03"'; sleep 8; } | telnet esp3d 8888
{ perl -e' print "\x00"'; sleep 8; } | telnet esp3d 8888
{ perl -e' print "just some simple-text with no newline at the end"'; sleep 8; } | telnet esp3d 8888
```

-------

It seems that the printer firmware will halt/wait to parse a statement until it has received the newline '\n' character. **Any of the following statements work without issue / will resume printing** because they send the newline character.
```
{ perl -e' print "\n"'; sleep 8; } | telnet esp3d 8888
{ perl -e' print "\x03\n"'; sleep 8; } | telnet esp3d 8888 
{ perl -e' print "just some simple-text with a newline char at the end\n"'; sleep 8; } | telnet esp3d 8888
```

---------

Issue first discovered as part of [ESP3D Issue #297](https://github.com/luc-github/ESP3D/issues/297)
Might be related to Prusa-Firmware Issue #1394

AnHardt, 2019-02-07:
Per definition a g-code statement ends with a '\n'.
How else would you be able to decide if v.e. a "G28" or a "G2" was wanted to be send.

AustinSaintAubin, 2019-02-08:
I am aware that a newline statement is required... 
The problem is that the printers and seem to halt while waiting on the newline statement.
Might be a good idea to process the Raspberry PI / USB Serial port separately from the controller on the front of the printer.

There is also no timeout for at statement that does not end with newline. 
This means, for example, that if there is noise on the serial rx line the printer could halt indefinitely... while still holding temp....

closed 2023-09-19 by github-actions[bot]