prusa3d/Prusa-Firmware #1516: 3.5.2-RC1 - Print fan PWM bursts

Dolnor, 2019-02-08:
With the change to soft PWM now the fan is indeed inaudible, however if running at a fraction of the full speed, the fan will now be on a weird cycle where it would periodically burst with a faster PWM signal resulting in an audible blower pitch change and an apparent air output increase. 

For something like ABS, this is probably going to result in layer separation as the fan output is not consistent and occasionally, there will be more cooling done than requested.

PavelSindler, 2019-02-08:
Thank you for feedback. We will try to improve this behaviour.

stahlfabrik, 2019-02-10:
Same issue here. I set the fan to M106 S125. Roughly every six seconds the fan turns up fast and then falls back to the slower speed.

NielschB, 2019-02-12:
Also same behavior here. When manually set fan speed from tune menu, every few seconds it increases speed en than returns back to lower speed.

zoltanf, 2019-02-15:
It's true, it's clearly audible. However, this is nothing compared to #460, so I would rather keep it like is that going back to software whining :)

NielschB, 2019-02-15:
Its not the audible part that's the issue here, its about fan speed bursts that can cause layer separation and deformations due to the higher amount of airflow.

zoltanf, 2019-02-15:
I agree, however I did link to #460 explicitly because this problem was introduced while solving that one. Another questions is if this is significant. Did someone try to measure actually the increase in burst rpm?

HolgerHU, 2019-02-22:
FW 3.5.2 - As already mentioned, the speed of the parts fan pulses within a few seconds. Furthermore, the rotation speed can only be adjusted in rough steps. But we need a uniform adjustability of dig. 0..255 with a step size of 1 dig. PWM. It seems you are not quite sure how important this fan is?
The PWM noise is unimportant compared to the meaning of this fan. One way to switch the operating mode of the fan would be useful in the settings of the printer and can solve the issue for different applications. Thx. We would therefore like to keep the fw 3.5.1.

ra100, 2019-02-23:
What I've noticed is that thanks to this bursts nozzle temperature is oscilating +/- 1 deg of target temperature, which can't be good for print.

F0x06, 2019-03-04:
Any solutions for this issue ?

stahlfabrik, 2019-03-04:
I would also be interested in having this fixed. Makes pwm with duty cycles < 100 pointless. More silent but unusable.

rennne, 2019-03-11:
Does this also occur at 30% fan speed ?

F0x06, 2019-03-11:
Im printing PETG right now, fan speed is 153, target temp is 235. I see oscillations of 2 degrees, 234 - 236, this has clearly an effect, i can ear the fan bursts also... never had this issues before this bug was introduced, i'll tell later if this has an impact on the quality

haarp, 2019-07-15:
This might be intentional. A regular pulse at 100% fan speed in order to reliably read the tacho signal, which is impossible while PWM is active.

Also see https://github.com/gregsaun/bear_extruder_and_x_axis/issues/20#issuecomment-480524299

juckendesAuge, 2022-04-16:
@3d-gussner 

I looked for it, but I didn't find it.
After all, this is always described differently. That it is because of the PWM detection, I also did not know.

Now, in any case, the PID tuning no longer makes sense.
I have made before the pressure two times a PID tuning to 240°C. After the first time let cool down completely and then again 240°C PID tuning.
The nozzle temperature fluctuates since the fan is accelerated in the interval every 6 seconds to 100% extremely.

I have recorded a video about it.

Many greetings

juck.

Translated with www.DeepL.com/Translator (free version

https://user-images.githubusercontent.com/24550429/163690223-480bc6cc-1c70-4f25-8dee-ecf9029b9949.mp4

)

Prusa-Support, 2022-05-13:
Hello! Sorry for the late response to this thread. The print fan PWM bursts are normal and intended: the fans generate at low RPM so much noise on the signal line that it is hard to distinguish if it is spinning or not, to be able to get some "readable" values the fan is forced to spin up. 

@juckendesAuge from this specific video I don't notice anything too irregular. Very slight fluctuations are still regular - are you sure, however, that there was a drop specifically due to print fan speed increase? In that case it could be about positioning of the fan shroud, as it's quite easy to "over-tighten" the bolt holding it, resulting in the fan shroud "at an angle", so cool air goes onto the block, while it shouldn't really. If still relevant, feel free to contact me directly via email info@prusa3d.com mentioning me in the subject ("attn: Alessandro") and we'll look into it from there. 

Closing the issue for now as the initial report was for regular, intended behavior. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-13 by Prusa-Support