prusa3d/Prusa-Firmware #1521: Pause during print for heating

korang, 2019-02-10:
I am using the newest firmware(3.5.1) on my MK3.  During printing I am getting a pause then the display is saying heating...  then the print resumes.  I have never seen this behavior before. Here is a link to video showing it happening.   https://youtu.be/qYR3xPORSD4

uniacke1, 2019-02-10:
I am getting this as well but my print fails immediately afterwards.

F0x06, 2019-02-11:
I already had the exact same issue, but not on every prints, after the print resumes correctly, possibly due to temp decrease when part cooling fan is switched to full speed after first layer.

korang, 2019-02-11:
@F0x06 This happens throughout the print for me.  Not just on layer 2.

mmcglumphy, 2019-02-13:
I just commented in this thread about the same issues I am having. These seem to be the same problem. Try disabling the filament sensor.

https://github.com/prusa3d/Prusa-Firmware/issues/545

NielschB, 2019-02-13:
Same behavior observed in 3.5.2 RC1 when printing from SD. Seems like this happens at "random" times during the print. (PLA, fan at constant max) Strange thing is that the display shows the correct temperatures at the pausing moment, only once observed nozzle temp 1 degreeC too low, which in my opinion is no reason to pause, because the pausing adds artifacts to the print.

andredigenova, 2019-02-15:
Same issue, 3.5.2 final

PavelSindler, 2019-02-18:
If printer (MK3, MK2.5) )detects possible filament runout, it pauses print immediately and move filament back and forth and measures response. If it decides that filament is still there, print is then resumed and "Heating..." will appear for a while. 
Cleaning your sensor may help in some cases.

korang, 2019-02-18:
I do not see the filament moving at all when this happens.

Crunch69, 2019-03-03:
I now have the same issue.  the only way to print now is to turn the sensor off

MoshiBin, 2019-03-27:
Same here. Seems like it's more prone to happen in large solid infill areas - I'm guessing that the filament moves in such a uniform way that it registers as a clog.

aequitas, 2019-04-07:
Got this issue as well with 3.7.0-RC1-2175

leptun, 2019-04-07:
Also got this problem on mk3 with 3.7.0 when printing with prusa black petg

kyjak, 2019-05-11:
Same problem, the printer does not resume at the same position. After heating pause, I get a layer shift.

kyjak, 2019-05-11:
> 
> 
> If printer (MK3, MK2.5) )detects possible filament runout, it pauses print immediately and move filament back and forth and measures response. If it decides that filament is still there, print is then resumed and "Heating..." will appear for a while.
> Cleaning your sensor may help in some cases.

Why not add a proper message instead of the misleading "heating ...".
Also, I get a layer shift after resume. This needs a fix, ruins prints multiple times.

acid-zero, 2019-06-13:
Just got this same issue on my newly built mk3s with firmware 3.7.1
Printing thingiverse model 2806295. Printer paused 3 times while printing the long/straight section, just below the 3D PRINTER TEST words. Each time extruder moved to front left corner, got the Heating message for a couple seconds, printing resumed but with a layer shift in the X-Axis.
Print still going and, so far, no more pauses.

Ma-Ghasemieh, 2019-10-24:
same issue

closed 2023-09-19 by github-actions[bot]