prusa3d/Prusa-Firmware #1522: Print fails after "heating" (Axis crash & crash detection does not recover properly)

uniacke1, 2019-02-10:
Just had a larger print fail in the same place twice. First all movement stops and the screen displays "heating", just like in #1521. Then when it resumes it just wanders off the print bed and crashes the axes. 

I've verified the bug is not Slic3r as the gcode from both the release and current alpha are identical barring comment changes.

The offending G-code, STL, and some images are attached. The unusual thing is the print right before this was very similar in nature, but completed without issue. Failure happens on the second layer just as it gets close to the front right corner.

First image is the first attempt, was in silent mode so no crash detection. Second image was in normal mode and detected the crash but completely failed to resume correctly. Belts are good and tight so it is unlikely they are slipping during the crash.

[Backtray.zip](https://github.com/prusa3d/Prusa-Firmware/files/2849382/Backtray.zip)
[image]
[image]

uniacke1, 2019-02-10:
Update: Just had it happen on the same gcode file (but different place) in 3.5.1. So there is something about this print that exposes a firmware glitch or problem with my printer.

uniacke1, 2019-02-11:
Having thought about this some and considering others have seen heating without failure, I wonder if it re-executes the last G-code it was partway through when resuming. So if your print has lots of small steps and direction changes, it's not noticeable, but if it happens during a long straight shot (like my infill above) it makes the nozzle just wander off as it tries to re-do the part of the line it already completed.

AnHardt, 2019-02-11:
That's very likely.
Haven't analysed Prusa-Firmware/Marlin cloth enough to be definitiv, but i do know MarlinFirrmware/Marlin.
There, not only the current move is lost, but the complete planner-buffer.

andredigenova, 2019-02-15:
I have this issue as well. Printing a large model in PLA and the print head just stops out of nowhere and the display says "Heating". According to the temperature displayed, the temps are fine and it doesn't need to adjust. I'm running 3.5.2 final release.

inquisitiveKangaroo, 2019-02-18:
Hi,
thank you for the report, we appreciate it.
related with
**Pause during print for heating #1521
3.5.2 heat bed temp fluctuations #1542**

ezeackermann, 2019-03-26:
the only firmware that work fine is 3.5.1. The next releases have the temperature fluctuation. Also I see that the heated bed are more time on with the latest releases. I see this in the powermeter of my UPS.

tiboine, 2019-07-08:
I had this problem with FW 3.7 (MK3). 
How is FW 3.7.2?

ezeackermann, 2019-07-08:
I have the same issue when I print across all the bed (from left bottom corner to right up corner) 45 degrees. I think that is a problem with the calibration or with the limits of the bed. Also in the last print I receive the Heating message.  Is true that here in Argentina this days are very cold, I don’t know if the room temperature have anything to do with it

uniacke1, 2019-07-08:
I've not seen it anymore in 3.7.2 but in all fairness that's not conclusive - I disabled my filament sensor at the time and only re-enabled it now I'm running a skelestruder with redundant sensing (reads both filament and bearing. Bearing-only indirect mods would still give me false runouts).

kazibole, 2019-08-08:
I just experienced this today on an MK3 running firmware 3.7.2. The issue appears to be related to the filament sensor recovery. The serial terminal showed that the filament sensor gets triggered (this is when the LCD screen shows "Heating..."), performs the filament test, then recovers. However when it recovers it continues straight past where the print should end and hits the maximum build plate area, triggering a crash.

jelson, 2019-09-08:
+1 -- I am consistently seeing this failure. The offending print looks very much like the one pictured above, i.e. a large rectangular area that nearly fills the print bed with 45-degree-angle fill. Several times during the 2nd layer, the printer pauses with "Heating...", and when it resumes it runs off the edge of the bed and crashes the head. This is with a Prusa i3 mk3, firmware 3.7.2-2363 and gcode from PrusaSlicer-2.0.0 win64-201905201652.

wavexx, 2019-11-29:
@uniacke1 @kazibole A relevant PR which should fix the crashes has been merged. Are you able to test a build of the current FW in order to test it?

You might still see the filament sensor leaving marks, but it shouldn't crash anymore.

uniacke1, 2019-11-29:
Sorry, wish I could help but I've since upgraded to the MMU2S which changed to an IR sensor.

kazibole, 2019-11-29:
@wavexx I have since upgraded my MK3 to MK3S, so I don't know if I would be able to reproduce the issue.

albjerre, 2020-01-20:
I have just had the same issue three times in a row. I re-sliced the print in between print 1 and 2. But the results were the same. The display didn't say "reheating" though as some other have mentioned.  The print was fine until midway on the second layer. Then the printer suddenly stopped for a few seconds before moving on. It made the inconsistent lines you can see on the photo. It did that two or three times, until it at the last stopped suddenly moved very fast out to the edge of the build plate and crashed.
I am using Prusa Mk3, firmware 3.8.1, PrusaSlicer 2.2.9, Spectrum PETG filament, 0.2mm quality profile.
[image]

wavexx, 2020-01-27:
Could you try with the latest 3.9.0 beta firmware?

albjerre, 2020-01-28:
Turning off the filament sensor solved the problem for me.

wavexx, 2020-01-28:
That's one solution, although we'd like to fix with *with* the filament sensor on :)

acekrystal, 2020-02-25:
I have the same problem. I had the "heating" and pausing while printing already for a long time. But this is the first time I see him totally crashing into the sides of the max travel distance after continuing, or continue printing in totally the wrong places.

albjerre, 2020-02-28:
@wavexx Yes, of course that would be better. Unfortunately I just don't have the time to fiddle with beta firmware.

Jugsy, 2020-03-15:
Experiencing this exact issue on MK3 with 3.8.1

I'm trying with the filament sensor off to confirm, but I'll also try 3.9.0 with fil sensor on and report back

[image]

closed 2023-09-19 by github-actions[bot]