prusa3d/Prusa-Firmware #1523: Invert Stepper - Bug

TigerClaw33, 2019-02-11:
Hi, 

I have a question. I use a Prusa Mk3 Mod with inverted y-Stepper. I changed the value in the firmware but it seem's that there is  bug with calibration with changed values (inverted) for steppers. 

In the zaribo 320 v3.5.0-RC1 there is a bugfix for that but not in the Prusa Firmware.

Panayiotis-git, 2019-02-11:
IIRC, there exists a push request with a fix for this issue, waiting approval for a very long time.

Dolnor, 2019-02-12:
This is still sitting unmerged unfortunately :/ 
https://github.com/prusa3d/Prusa-Firmware/pull/1263

3d-gussner, 2019-10-08:
@inquisitiveKangaroo @DRracer This issue is taged as `help wanted` but there is a PR https://github.com/prusa3d/Prusa-Firmware/pull/1263 since October 2018 waiting to be pulled.

cimoalpacino, 2019-10-08:
#1263 is very nice in my opinion.
I use it on my prusa build made for ramps board and it works great.

leptun, 2019-10-08:
@DRracer I also use #1263 in the MKP branch.

closed 2023-09-19 by github-actions[bot]