prusa3d/Prusa-Firmware #1526: LCD hieroglyphs

TigerClaw33, 2019-02-12:
Hi,
I built anf flashed the hardware exactly how described but after the first message "XYZ Research and Developement" I can only see lcd hieroglyphs in all variations. Do someone know why?

Barnabasek, 2019-02-12:
From the compiling instructions - "In file "Firmware/config.h" set LANG_MODE to 0”. Then it should work.

Prusa-Support, 2022-05-12:
Hello! Sorry for the absolutely late response here. Was it with one of our original firmware versions, or anything different? Is the issue still relevant? In case of anything more to review, feel free to let me know at any time info@prusa3d.com mentioning me in the subject ("attn: Alessandro"), I'll be glad to help. 

Alessandro Pantaleo
Prusa Research

closed 2023-09-19 by github-actions[bot]