prusa3d/Prusa-Firmware #1531: Pause print is now BROKEN with 3.5.2.

zkirkpp, 2019-02-13:
WARNING:

Since updating to the 3.5.2 firmware yesterday the Pause function is screwed up. I hit pause 7 hours into a print and it turns off the nozzle heater and CONTINUES PRINTING. Luckily I was paying attention and stopped the print before it started grinding the filament as the temp dropped. To make matters worse this was a ton of Prusament that is now wasted.

inquisitiveKangaroo, 2019-02-13:
Hi, thank you for your feedback. May I ask you what printer type you are using?

zkirkpp, 2019-02-13:
> Hi, thank you for your feedback. May I ask you what printer type you are using?

Printing on an MK3 purchased May 2018.

inquisitiveKangaroo, 2019-02-13:
Thank you, we will look on it. It sounds like high priority issue.

inquisitiveKangaroo, 2019-02-15:
Hi,
The pause function works fine to me (tried on multiple printers). Could you provide us more info such as the g-code or even better a log where this issue appeared?

inquisitiveKangaroo, 2019-02-15:
Maybe contact the customer support to exchange your Einsy for new one. We'd like to make further tests on yours.

zkirkpp, 2019-02-15:
I reflashed the firmware and reset the printer to factory defaults and the
issue is resolved.

3d-gussner, 2019-02-15:
@zkirkpp If that issue is resolved you can close the issue yourself to clean up the issue list

closed 2019-02-15 by zkirkpp