prusa3d/Prusa-Firmware #1532: Filament autoload works with nozzle not heating

mark0815, 2019-02-13:
Just installed 3.5.2 on my MK3 and did the following:

1. Select unload filament from the menu
2. Printer asked for filament type, selected PET
3. Nozzle heated up, when the target temperature was reached the Printer asked to hit the knob to unload filament. Filament got unloaded, nozzle heating turns off.
4. Inserted new Filament, autoload kicks in and loaded the filament, without enabling the heating or asking for preheat.

I did the filament change quite quickly so the nozzle still had around 190 degree C during filament load.

workinghard, 2019-02-13:
Loading is allowed if the nozzle is (i think 180C, but not sure). It doesn't matter if Autoload or manual load.

mark0815, 2019-02-14:
Ok. I should have read the manual ;-)

closed 2019-02-14 by mark0815