prusa3d/Prusa-Firmware #1534: Issue with MK3 3.5.2 and MMU 1.0.3

Raekro, 2019-02-14:
Hello,
I have a problem with an MK3 firmware 3.5.2 and an MMU 1.0.3. At the first filament change, at the time of ejection, the printer restarts. The problem does not occur with the combination 3.5.1 and 1.0.3 and the same Gcode file. 
I asked support if this was normal, they told me that no, the MMU in 1.0.3 is supposed to work with 3.5.2, and asked me to make a bug report here.
Tested: 
- Factory Reset MK3 : No ok
- Factory Reset MMU : No ok
- Downgrade 3.5.2 -> 3.5.1 MK3 : Ok

I still think that the 3.5.2 firmware is not functional with the MMU in 1.0.3. 

Is an update of the MMU planned? Have any of you encountered the same problem?

PavelSindler, 2019-02-14:
We will look on it.

Raekro, 2019-02-14:
> We will look on it.

Ok, Here is what happens on the LCD screen when the problem occurs :
[Issue#1534.mp4.zip](https://github.com/prusa3d/Prusa-Firmware/files/2865053/Issue.1534.mp4.zip)

JohnnyDeer, 2021-01-19:
Hi @Raekro, is this still a issue on newest FW versions?

Raekro, 2021-02-09:
> Hi @Raekro, is this still a issue on newest FW versions?

No problem since 3.6. Sorry for the inconvenience, I forget to close this report.

closed 2021-02-09 by Raekro

JohnnyDeer, 2021-02-09:
@Raekro thank you for update.