prusa3d/Prusa-Firmware #1535: Prusa MK2.5 Serial Port 2

dronecz, 2019-02-14:
Hi, 

we are developing access system for Prusa printers, which we have here in FabLab Brno, and so far it is working with no problem on MK3 printers. We use MMU2 serial port or RPi port, if there is MMU2 on printer, but if we want to use it on MK2.5 printers the response is quite different. 

Is there somewhere more info about difference between MK2.5 and MK3 use of serial ports. I know there is HW difference between miniRambo and Einsy board, but I thought that FW will use same functions when is same for MK3 & MK2.5  

Thx for answer.

3d-gussner, 2019-02-15:
Hi,

The Atmega 2560 has by default 4x serial TX and RX (0-3).
- The TX0/RX0 pins are used to communicate with the Atmega53u2, which handles the USB communication on both boards.
- The TX2/RX2 pins are connected to P3 pins 3/5 (expansion port) on both boards. This is used to communicate to the MMU2 via serial on both boards.
- The TX3/RX3 pins are connected to P2 pins 5/7 (LCD 2 port) on both boards. 
- The TX1/RX1 pins are connected on the miniRAMbo to P1 pins 3/5 (LCD 1 port) BUT on the EINSY these are connected to rPIport pins 6/8 to enable an additional serial communication.
- The Einsy uses other pins to P1 pins 3/5 (LCD 1 port) to support the LCD screen.

So you see that the miniRambo has only free 2x serial ports to communicate in the Prusa MK2.5 build. 1x USB (via the 32u2) and 1x at P3.

The Einsy on the other side has 3x serial ports to communicate, 1x USB (via 32u2), 1x at P3 and 1x at rPIport.

I guess Prusa tweaked/optimized the firmware on the rPIport and P3 port for their needs: 
- rPIport for 3rd party serial communication 
- P3 port optimized to the MMU2 communication needs and so maybe bit different to the usual serial communication 
- USB serial communication to 3rd party products on both boards.

You could do some fancy hardware stuff on the MK2.5 + some additional firmware tweaking to get an additional serial port.

Hope that helps a bit.

dronecz, 2019-02-15:
Thanks @3d-gussner for description. 

I´m just wondering what is the deference? FW is same, difference is just in variant file and it is behave different. 

I was looking at MMU2 FW and I saw that begin function is called like this: 
`Serial1.begin(UART1_BDR, SERIAL_8N2); //serial1`

so I changed my begin function and now I getting this: 

[image]

which at the end contain what I need from FW, but it will not accecpt `start` to continue communicate. 

Any thoughts  on that? 

Thanks

3d-gussner, 2019-02-15:
@dronecz 
The Firmware is the same, but depending what you choose code is used/compiled or not. So if you have an MK3 it will add the TX1/RX1 communication and on both the TX2/RX2 is optimized for the MMU2 i guess.
Sorry i din't dive that much in the serial code, so can't give you the right hint to solve the issue.

Let's hope Prusa or somebody else can make it more clear.

dronecz, 2019-03-07:
@PavelSindler or @mkbel could you please reply at least with some information?

closed 2023-09-19 by github-actions[bot]