prusa3d/Prusa-Firmware #1536: Feature Request: Auto Print from File

assumptionsoup, 2019-02-14:
I've noticed that for most of my prints I'm always printing the latest file on the SD card. It would be nice if there was an option in the firmware to automatically print a file with a specific name on startup.  

Something that would be hard to accidentally discover, like `__auto_print__.gcode`.  On Startup / SD card insertion the printer could ask:
```
Do you want to start the automatic print file __auto_print__.gcode now?
[YES]   [NO]
```

Or another way to implement this could be a "Quick Print" option in the main menu to print the last modified file.
```
...
Print from SD card
Quick Print
...
```
That prompts:
```
Do you want to print awesome_print_v12.gcode now?
[YES]   [NO]
```
For me, this feature would reduce the amount of time I spend twiddling with the printer to do the same thing, and could reduce mis-clicked files due to the shoddy encoder.

inquisitiveKangaroo, 2019-02-15:
Hi,
thank you for an interesting idea. However, if i understood you well I do not think the outcome would be much beneficial.
Let me explain: Right now (on FW: 3.5.2) when you insert the SD card, you get directly into the SDcard menu. Requiring to rotate the knob one step down a press it.
In case of implementing your feature you would get a message after inserting an SD card: 
'_Do you want to start the automatic print file auto_print.gcode now?
[**YES**]   [**NO**]_'
Probably set to 'no' by default. Which mean to start print you need to rotate the knob one step up a press it.
In case you want to print the first gcode it sounds quite the same to me, however, in case you do not, it would take you at least two extra clicks.

assumptionsoup, 2019-02-15:
Ah, I hadn't realized that the newest firmware jumps you straight to the SD card menu. You're correct, that this pretty much a moot point, as long as you're asking for confirmation.  A minor correction, is that you still have more steps to perform if you have folders on your SD card, as those show up first, and you have to scroll past them. But it's a small thing to delete those.

Daverkex, 2019-05-22:
Duplicate from: #1610

assumptionsoup, 2019-05-22:
@Daverkex  I agree. I'll close this issue. I like where the discussion is going in the other one.

closed 2019-05-22 by assumptionsoup