prusa3d/Prusa-Firmware #1537: WHY another MK3S branch?

3d-gussner, 2019-02-15:
Why another github branch? I hope this is temporary!

We saw what happend when the MK3 branch was created from the MK2 branch, and Prusa started to develop in both separately. 
- Nearly all of fixes/pull request (also from the community) to the MK2 branch during their existence next to each other never made it into the MK3 branch.
- After MK3 branch got public: Old Fixes HAD to be manually migrated one by one to get the MK3 branch firmware as stable the MK2 was at that time.
- The MK2 Firmware isn't maintained anymore, the last official hex file by Prusa for the MK2 is from 12 November 2017.
  - even the quite important M110-fix which was pulled on 22 December 2017 was never provided as a hex file by Prusa.

**Please don't make the same mistake again**

PavelSindler, 2019-02-15:
This branch is temporary just for initial release.

3d-gussner, 2019-02-15:
Sorry, was just bit worried as the MK3S 3.5.3 hex file looked like it was created from that branch

closed 2019-02-15 by 3d-gussner