prusa3d/Prusa-Firmware #1542: 3.5.2 heat bed temp dips

dryja123, 2019-02-15:
Hello,

I flashed the official release of 3.5.2 and started to observe roughly 7 degree fluctuations in bed temperature in a patterned amount of time.  I flashed back to 3.5.1 and that resolved the issue.  

Screenshot with 3.5.2 firmware:
[image]

Screenshot with 3.5.1 firmware
[image]

I did not PID tune the bed because the temps were consistently stable before I used this firmware.  Also, my printer is kept in an enclosure.

bobek, 2019-02-16:
I do see relatively frequent pauses mid-print (e.g. about every 10-15 minutes) after upgrade to `3.5.2` on MK3. 

I was under impression that it was caused by nozzle temp as that was the one under target and message was generic "printer heating" which normally refers to hot-end (all based on quick observation of LCD when printer paused). Unfortunately I don't have temp graphs to verify this :(

Everything worked without an issue on `3.5.1`, nothing else has changed.

PavelSindler, 2019-02-18:
@dryja123 : Thank ypu for reporting this issue. We are currently testing it.
@bobek these pauses are probably related to this feature:
If printer detects possible filament runout, it pauses print immediately and move filament back and forth and measures response. If it decides that filament is still there, print is then resumed and "Heating..." will appear for a while.
Cleaning your sensor may help in some cases.

aemileski, 2019-03-01:
I thought the change was an increase to the bed PWM frequency.

I would definitely recommend a PID calibration for that, as it does have significant impact.

ezeackermann, 2019-03-26:
the only firmware that work fine is 3.5.1. The next releases have the temperature fluctuation. Also I see that the heated bed are more time on with the latest releases. I see this in the powermeter of my UPS.
Downgrade the machine and you will see the difference.

closed 2023-09-19 by github-actions[bot]