prusa3d/Prusa-Firmware #1545: M190 (wait for bed temp) overwrites set bed temperature

rennne, 2019-02-16:
I am working on an improved Start G-code, which start printing faster.

Therefore I want the bed to heat up to eg. 60 but only wait for 55 before continuing.
>M140 S60 ; set bed temp
>M190 S55 ; wait for bed temp

Unfortunately M190 sets the bed temperature back to 55.

The biggest issue with that is that the pid controller tries to reach 55 exactly, therefore it reaches it later than, when it would be set to 60.

For now I work with a workaround, but it is slower than it could be:

>M83  ; extruder relative mode
>M104 S170 ; preheat extruder without oozing
>M140 S[first_layer_bed_temperature] ; set bed temp
>M190 S{first_layer_bed_temperature[0]-5} ; wait for bed temp
>M140 S[first_layer_bed_temperature] ; set bed temp again, needed because of firmware bug
>M104 S205; increase extruder temperture further during homing
>G28 W ; home all without mesh bed level
>...

haarp, 2019-02-24:
Why not swap the commands?

    M190 S55 ; wait for bed to reach 55
    M140 S60 ; start heating to 60 and continue

rennne, 2019-02-26:
That is my workaround anyway, but it takes unnecessary extra time, and it is the wrong behavior:
[https://reprap.org/wiki/G-code#M190:_Wait_for_bed_temperature_to_reach_target_temp](https://reprap.org/wiki/G-code#M190:_Wait_for_bed_temperature_to_reach_target_temp)

aemileski, 2019-03-01:
M190 is working as designed.  It always sets the bed temperature.

M190:  set temperature and wait    
M140:  set temperature and continue

rennne, 2019-03-02:
Ok. I interpreted that wrong.

How do I wait for the bed temperature without setting?

closed 2023-09-19 by github-actions[bot]