prusa3d/Prusa-Firmware #1546: FW 3.5.2 - Print Fan Error

Hotpocketdeath, 2019-02-16:
All my fans are working and I never had an issue with them until after I updated to 3.5.2.

After the update, Soon after the print fan turns on (4th layer PETG), the print will stop with a Print Fan Error.  But the fan is still working and air is flowing.

However, I am not using the stock Prusa blower fan.  A while back, it began to fail, so I purchased some Zaribo fan replacements.  There was no issue using this fan before updating and the printer works fine if I disable fan checking.

jdn-za, 2019-02-19:
Same issue

jdn-za, 2019-02-19:
So my issue was resolved by switching from using 2 fans to 1 fan (yes I had modded it to use 2)

I guess the new firmware is more sensitive to pwn pulses?

oxivanisher, 2019-02-23:
I also had this error just now. Multiple times but with no fan mod or anything. Also with PETG and on a two-day-print at 46% (so it worked for almost 20h and then suddenly this error).

oxivanisher, 2019-02-26:
> I also had this error just now. Multiple times but with no fan mod or anything. Also with PETG and on a two-day-print at 46% (so it worked for almost 20h and then suddenly this error).

My report could be false, since there is probably a broken wire on my circular fan. Once I have it replaced, I can say more.

JPTa, 2019-04-05:
Same issue with 3.7.0. Same blower fan as Hotpocketdeath is using.

koenner, 2019-11-02:
Could be coincidence with the firmware upgrade, but I had [this issue](https://www.youtube.com/watch?v=cbKcj5tHjP0).  Just want to share since it took a while to root cause this failure mode.  Also, note that the Prusa customer service in dealing with this issue is total garbage - not impressed.

closed 2023-09-19 by github-actions[bot]