prusa3d/Prusa-Firmware #1548: MK2.5 FW 3.5.2 MINTEMP error

petrzelka, 2019-02-17:
Since the upgrade from 3.5.1 to 3.5.2, I'm getting MINTEMP errors after the start of print.
My printer is placed in a garage (in an enclosure) currently at ~12C. In previous FWs, when printer booted and the sensors were reading less than 15C I had to use a hairdryer or something similar to get the bed and nozzle over 15C and reboot. That sucks, but well, okay.
But now with 3.5.2, it normally boots even at 12C, I can start the print, heating begins and I receive a MINTEMP error about a 1 minute later, when nozzle is at ~100C and bed at ~25C. That doesn't really make sense, does it?
Let me know if you need any details or exact timings/measurements of the events described. Thanks.

jsilva-iii, 2019-02-18:
I too am experiencing the same error - my printer is inside and room temperature is 20C - Also getting Min temp error since the 3.5.1 - never experienced it before. I have to reset the printer and it is fine.

SHolman77, 2019-02-19:
I am having the same problem as described above. After resetting it is fine, very annoying error tbh.

PavelSindler, 2019-02-19:
Hi, we guarantee working temperature range: 18 °C (PLA)-38 °C, indoor use only. 
If you will use printer in cold environment, you will encounter this issue. However if you reset printer and start print again, it should pass.

jsilva-iii, 2019-02-19:
Pavel

Hi, this behaviour has been absent up until the new firmware - if I
downgrade the firmware, no problem. The printer is inside my apartment with
temperature at 19C.

Thanks
Jorge

helixios, 2019-02-20:
I'm getting this occasionally as well

SHolman77, 2019-02-21:
> Hi, we guarantee working temperature range: 18 °C (PLA)-38 °C, indoor use only.
> If you will use printer in cold environment, you will encounter this issue. However if you reset printer and start print again, it should pass.

My printer is inside my apartment, it is 21 degrees when it starts the preheat. 
I never had this issue before when preheating. I did however had issues with the previous firmware when I could not print ABS anymore because of the same error but mid print. (known issue)

Zonda95, 2019-02-27:
I have also had this issue since the firmware update. Looking at the firmware the mintemp it will check for is 30°C, it has a delay timer but I'm not exactly sure how that works with the "short timer". In practice it seems to check for the min temp after either 30 or 45 seconds so I have just been turning on my bed via octoprint for 30 seconds so that it will reach 27°C (~24°C ambient) and then I can print the file with no issue as it will be above the threshold when it checks. 
Today I added in gcode to my printer profile to do this for me. It turns on the bed to 35°C waits 25seconds then turns the bed off and then waits a little more and starts the print. There is probably a "better" option but this seems like it should work.

franzbischoff, 2019-03-06:
I get the MINTEMP error **every time** I start the first print (Cold printer). But I can avoid the error if I tell the printer to heat the bed to 30ºC, and after some seconds, to 31ºC, 32ºC, etc, until it gets over 30ºC, so I start to print and leave it. Seems to be a tight threshold.

Printer: mk2.5, firmware 3.6.0-2069, roomtemp 20ºC.

SkavenKF, 2019-03-08:
I have the same behavior with identical settings as franzbischoff

cimoalpacino, 2019-03-09:
In fw 3.1.6 the mintemps were defines 15C lower for bed and extruder than in version 3.6.0.
FW 3.1.6:
#define HEATER_0_MINTEMP 15
#define BED_MINTEMP 15

FW 3.6.0:
#define HEATER_0_MINTEMP 30
#define BED_MINTEMP 30

Though in both versions the ambient mintemp remained the same (25). 
Could maybe this cause problems?

franzbischoff, 2019-03-10:
I guess that is the problem @cimoalpacino; the bed may not reach the 30ºC fast enough not to trig the MINTEMP ERROR. MK3 may be ok because of the 24V or the Ambient thermometer; I don't know.

cimoalpacino, 2019-03-10:
Could you try to reflash your printer with adjusted temps to 15-20C and see if the problem persists or is gone?

marccobe, 2019-05-02:
same issue here. mine is at 20C normally. 
I don't see the point that the mintemp is 30 degrees now... half countries / half year we are below that temperature

chatrat12, 2019-08-25:
Been getting this error for a few months now. I have to restart the print and it will work fine. Can you have 2 separate defines for 12v beds and 24v beds? This issue isn't a show stopper. But it is really annoying when starting up a print.

Prusa-Support, 2022-05-11:
Hello! Sorry for not answering on this thread before. The thresholds for Mintemp have changed over time, at this time we have: 

MK2/S & MK2.5/S
Hotend thermistor - temperatures below 22 °C (71.6 °F)
Heatbed thermistor - temperatures below 15 °C (59 °F)

MK3 & MK3S & MK3S+
Hotend thermistor - temperatures below 10 °C (50 °F)
Heatbed thermistor - temperatures below 10 °C (50 °F)

At the same time, let us keep in mind that printing in very cold environments, outside of the recommended operational temperature, can have negative effects due to other factors than the printer's electronics, like material properties. 

I'll close the issue for now as there were improvements and changes since the latest comments, in case it's still relevant do not hesitate to let me know at any time via email info@prusa3d.com mentioning me in the subject ("attn: Alessandro"). 

Alessandro Pantaleo
Prusa Research

closed 2022-05-11 by Prusa-Support