prusa3d/Prusa-Firmware #1551: Nozzle Max Temp error + bed Min Temp Error + Cooling Fan Hunts

jsilva-iii, 2019-02-18:
3 issues with FW 3.5.2 en_de on Mk 2.5, with MMU2 FW 1.0.4-RC1 Firmware for MMU2

1) Printer is kept indoors, at 19C, first print of day get Min temp error after attempting to print with PET - have to reset the printer and then it is fine
2) The cooling fan (radial one, for cooling prints) hunts, speed varies - you can hear it, this has been reported
3) This one is BAD! Printing a single filament, with nothing loaded, printer went on to do bed leveling, then went back to the 0,0 position and just hung, didn't prompt for any filament. I started smelling something burning (either plastic or electronics) - reset the printer and noted that the nozzle temperature was at 400C, the extruder fan started running at full, with the buzzer going - I got the MAX Temp error - never seen this one. I tried to re start the print and the printer wouldn't prompt for loading of the filament, I had to manually load a filament from the menu.

Prior to this firmware, I have never had any of these issues, I will probably revert to 3.5.1 - Issue #3 is pretty bad!!!

helixios, 2019-02-20:
I've been getting the mintemp error occasionally as well since updating the firmware to 3.5.2

aemileski, 2019-03-01:
If the temp reached 300C, the thermistor may be permanently damaged from the heat.  Prusa typically recommends 285C max, for a margin of safety (component tolerance).

I find thermal run-away happens above 260C if you don't calibrate the PID for that high of a temperature.

Also beware that at high temperatures, hardened steel nozzles can lose their temper.

closed 2023-09-19 by github-actions[bot]