prusa3d/Prusa-Firmware #1552: halts/heats up when not needed FW 3.5.2

port513, 2019-02-18:
I have a lot of weird problems with 3.5.2 at the moment.
Print halts then heat then continues again, when I look in the logs the temp have not changed more than +- 0.1C so why does this happens now?
It doesn't seems to make a bad print but every halt/heat cycle delays the print alot when printing over a few hours.

arcanemachine, 2019-02-19:
I had the same problem, but only after I switched to a 0.6 mm nozzle. After switching back to 0.4 mm, my problem went away.

Are you using a 0.6 mm nozzle by any chance? Were you having the same problems on 3.5.1?

kdomanski, 2019-02-22:
I am noticing similar issues. It happens rarely, but when the printer halts, the current temperature never differs by more than 1 degree from target.

F0x06, 2019-02-26:
Already had this issue in rare occasions, this can cause blobs and melt surfaces, seen in 3.5.1 and older

closed 2023-09-19 by github-actions[bot]