prusa3d/Prusa-Firmware #1555: [Feature Request] Detect extruder skipping and react on it

rennne, 2019-02-19:
I had prints where I had this extruder clicking (skipping thing) going on.

I am not sure if it the stepper motor which is unable to do a step or if the gear tooths are skipping over the filament

However some prints manged to get finished and others not.
From my observations I would conclude that, when it started clicking and it had a nice long perimeter to do afterwards it usually solved the issue and continued printing.
But if after a few clicks it did some retractions the extruder was just clogged and the print failed.

So i think it would be great that the printer could:
- Detect extruder skipping/clicking
and if so 
- Deactivate any retractions and deretractions and
- Increase the stepper current 

for a bit, maybe then 20mm of filament and afterwards go back to normal operation with lower current and retraction switch back on.

Further if it cannot be solved, the print could stop, that I can unload/load the filament and resume the print at the point where it started clicking

kdomanski, 2019-02-22:
@rennne I've also had extruded clicking with bad quality filament, which at times was as thick as 1.80mm. Upon closer inspection it was the idler pulley skipping gears, not the extruded motor. You might want to increase the idler springs tension, but better filament (i.e. with stricter diameter tolerances) will be spare you many hot-end clogs.

rambopierce, 2019-02-22:
I had repeated clogs and extruder clicking for a long time using multiple brands of filaments and got ‘lucky’ while researching one of them. I found a piece of filament broke off in the heater block that was 19.39mm long. The piece had a very small, but noticeable ridge at 12mm from the tip. That was the clue to me that the nozzle was not properly seated to the heat break. I disassembled the nozzle heater block, and heat sink/heat break and followed E3D’s assembly instructions with a couple of exceptions. They say to screw the nozzle all the way in, back the nozzle out 1/4 turn, and then screw the heat break in until they touch; I find that it’s a good idea to have it heated during this step as well as the final nozzle tightening step, with it heated the heat break screws in easier so you can tell when it has truly bottomed out against the nozzle. Also, I think it is best to back the nozzle out just slightly past 1/4 turn. Ever since getting this assembly error corrected haven’t had a single click or clog or MMU2 unload error.

There can be no leaking, even small, between the heat break and the nozzle.  Mine was so small that it was hard to find, filament was not leaking out onto the heat block where it would have been visible, but the gap was clearly a problem. I have seen many articles where folks have swapped out the heat break for another brand and then had great success. I have to wonder if their new found success is less about the other brand and more about they did a better job on assembly this time.

moojuiceuk, 2019-12-21:
I would love to see a skipping extruder check enabled in firmware too. Having just lost a print to this, it would be good if the firmware could detect the trinamic driver skipping steps on the Extruder motor. If there was say 3 skipped steps within 15 seconds, pause print, move hot end to the side and up, beep for attention and keep bed hot. Keep X, Y and Z steppers energised to prevent losing position, but power down Extruder stepper. Allow the user to do a quick cold pull then resume print? Would make sense.

Prusa-Support, 2022-05-20:
Thank you for sharing your ideas.

Unfortunately, the MK3/S/+ hardware doesn't generate sufficient data for detecting a non-moving filament with certainty, or reliably enough.
This issue is similar to issue #476, and it will therefore be closed as a duplicate.
Please check said issue and possibly refer to it in case of future discussions.

Moreover, this thread could be related to issue #765.
Please check this issue to learn more about our taken actions and our advice to possibly avoid the Extruder motor from skipping/clicking.
 
Michele Moramarco
Prusa Research

closed 2022-05-20 by Prusa-Support