prusa3d/Prusa-Firmware #1556: MK 3 Firmware 3.5.2 Software errors upon filament load

jhbinfla, 2019-02-19:
Hello:

Experienced a strange issue upon upgrade to Firmware 3.5.2 Version 1999 on a MK3 which does not seem covered already and seems to be a firmware glitch.  I was lucky enough to be close by and took pictures while it was happening which are attached below.

I have been running fine on all previous firmware versions and actually completed a print after updating to this firmware version.  Also, I noticed a couple of threads on the Prusa Forum with new users experiencing the extruder travelling to the right and crashing into the heatbed which is similar to this issue.

Downloaded and updated to Firmware 3.5.2 Ver 1999 and drivers 2.2.3 on 02/15/19 and successfully ran a print without issue.  Turned on printer today 02/19/19 and began heating to PETG to load filament.  Upon loading the filament with the menu load filament, I received the heating disabled by safety timer message #1 below.  It was not a long amount of time, started printer and then preheat to PETG temps so 5 - 10 Min max)  Screen would not respond after that (no picture).  I used the x button to restart, reheated to PETG and unloaded and reloaded filament.

While loading filament, the extruder then quickly traveled all the way to the right endstop (picture 3) and then I received the filament change message (picture 2).  I unloaded and reloaded the filament using the instructions on the screen (it was like a filament change, but unprompted). Then a crash detected message appeared on the screen (picture 4).  the bed moved all the way to the back Y position and extruder moved all the way to the left position (picture 5).  After this, the extruder was still heated, so I used cooldown to turn off heat and after cooling down I shut off the printer.

I am going to try again shortly and if it happens again, I will revert back to 3.5.1 or re download and reflash 3.52 unless there are other suggested testing procedures to try out.  My filament sensor was on and I just turned it off.  There was a filament runout and X crash detected on the last print (which I presume was the glitch I experienced as a print did not even start.

[image]
[image]
[image]
[image]
[image]
[image]

Forum posts of similar issues with losing home and x axis position

https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/calibration-problem-t27742.html
https://shop.prusa3d.com/forum/assembly-and-first-prints-troubleshooting-f62/printer-loosing-home-point--t27790.html
https://shop.prusa3d.com/forum/assembly-and-first-prints-troubleshooting-f62/mk3-loses-its-home-after-stopping-a-print-t17649.html

jbinfl, 2019-02-21:
Update:

I turned off the filament sensor and was able to complete 2 prints without issue...

Tomorrow, I will try again with the filament sensor on and see if the issue occurs again or not and update.

jbinfl, 2019-02-22:
Update 2:

I successfully loaded and printed 2 prints today after turning on the filament sensor.  Must have just been a strange glitch.  I will close the issue but will update if it comes back...

closed 2019-02-22 by jbinfl