prusa3d/Prusa-Firmware #1566: Bed leveling ends to low, cause the nozzle to be smuted with plastic

rennne, 2019-02-21:
During "G80 ; mesh bed leveling" the distance between the nozzle and the bed is at lowest around 1 mm. At the end it moves diagonally over the build platte and much more down than ever.

I assume it is 0.0 in z-direction. This cause that any molten rest of filament which was hanging on the nozzle during bed leveling to be squished onto the sides of the nozzle. Depending on the print-temperature this gets cooked there till it is black and hard or it falls of somewhen later during the print.
I consider both as not preferable.

I played a lot around with G-Code before and after, but I could not get the procedure to not doing that.

I think the most useful behavior in the long run, would be that G80 stops without the diagonal move at the end and this is done by G-Code instead.

But a quick short-term solution could be just:

- Move to 1.0 instead of 0.0 at the end.

aemileski, 2019-03-01:
Mesh bed levelling ends at 0.15 mm, which is the minimum layer height (to protect the PEI from excessive heat damage).

I recommend changing your Slic3r start-up code.

WARNING:  The following is off-the-top-of-my-head.  I'll post my actual code later when I get a chance.

Example:  After mesh-levelling and before printing the intro line

* Wipe 10mm to ensure the blob is behind the nozzle:  G0 X10.0
* Start the intro line (increase X by 10mm)
* Print the intro line (increase X by 10mm) 
* Retract and lift:  G0 E-0.8 Z0.8
* Big wipe, so ooze isn't carried into print area:  G0 X200.0
* Pause for half a second, to let ooze fall to bed:  G4 P500

Then add a un-retract to the before-layer code, something like:

{if (layer_num == 0)}G0 E0.8{endif}
{if (layer_num == 0)}G92 E0.0{endif}

rennne, 2019-03-02:
I tried a lot of G-Code variants kind of similar to what you are suggesting, but the movement to 0.15 at the end of G80 squishes sometimes just a bit of plastic onto the side of the nozzle. 
Depending on the amount which get squished at here it can take quite some time till it falls of, sometimes even a minute later, during printing.

I even go up 5mm to let it ooze there an then wipe sideways from there:

G80 ; mesh bed leveling
M104 S[first_layer_temperature] ; set extruder temp
G1 Z5
M109 S[first_layer_temperature] ; wait for extruder temp
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X30.0 Z0.2 E4.0  F1000.0 ; nozzle cleaning
G1 X60.0 E6.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0

It work really good, except when the plastic is squished on the side of the nozzle. Then all the wiping and everything is of only little use, because it only archives to clean the tip and not the sides of the nozzle.

The change to

- Move to 1.0 instead of 0.15 at the end.

would just prevent the nozzle form getting dirty on the sides in the first place

wavexx, 2020-01-27:
I don't think this is a FW issue as this is fully controlled by GCODE.
There is no perfect "purge line" I've seen so far :)

The thin/thick sequence which is used by default in the PrusaSlicer profiles is not bad. The "fat" portion of the line frequently drags away even the plastic which is stuck to the left side of the nozzle, which happens a lot when printing with PET. But the extruder needs to be low to work. Increasing the flow even more on the second can perform better depending on the material used, but you have to be careful: excess flow with ABS will do worse, so all in all it's a good average which I think one needs to tune.

One variant which I liked is doing a serpentine movement with 2x flow. This might be just slightly more successful at removing dangling stuff from the nozzle from the top/bottom sides as well

rennne, 2020-01-27:
It is the FW implementation of G80. It moves to 0/0/0 at the end, which causes the trouble.
There is no GCODE to prevent that move

wavexx, 2020-01-27:
Ooh, I get it now.

We cannot change this behavior by default as this would break existing gcode. Maybe adding a target coordinate (using X Y Z) to go to after leveling could be a good solution.

rennne, 2020-01-28:
Adding X Y Z for the move after leveling within G80 would be great. It would also offer the chance to prevent that diagonal move completely

closed 2023-09-19 by github-actions[bot]