prusa3d/Prusa-Firmware #1569: Request to be able to change the preset pre heat temps

port513, 2019-02-22:
Can we have the option to set the pre heat temps to custom settings?

My InnoPET prints really fine at 75C and 225/220C so there is no reason for me to heat to the normal PET temps.
As I understand now you have to recompile the FW to change these settings right?

tcurdt, 2019-02-25:
This's been bothering for a long time, too.

Especially now with the new upcoming filament load procedure this could be useful.
(Sorry, no - I don't want to heat the bed when changing filaments.)

inquisitiveKangaroo, 2019-02-26:
Hi,
you still have an option to preheat the nozzle for custom temperature by _Menu>Settings>Tenperature>Nozzle_. Also it is not nessesary to hit the ideal temperature for just loading the filament.
For ideal print temperature you can change it in slic3r.
However, I understand it would be cool to have a preset which only sets the nozzle temperature target.)

tcurdt, 2019-02-26:
@inquisitiveKangaroo Sure doable - but It's a whole lot of turning to get it to a pre-heat temperature that way. So thanks for acknowledging it as a feature request.

michalxfanta, 2019-10-11:
We are planning to add this feature to one of the next releases.

closed 2019-10-11 by michalxfanta