prusa3d/Prusa-Firmware #1571: MK3S + MMU2S - filament runout irrecoverable

DimitarKrastev, 2019-02-23:
Printer version: 3.6.0-RC1
MMU2S version: 1.0.5

Scenario:
When printing with i3 MK3S and MMU2S the FINDA detect no filament during print (filament runout).

Actions taken:
I removed the old filament from the PTFE and the extruder and I pressed the right button. The MMU ejects the filament, starts beeping and the printer says "Remove old filament and press the knob to start loading new filament." 

Issue:
The knob does nothing. Neither button on the MMU cause any effect. Basically there is no way of telling the printer you have actually changed the filament and the print cannot be resumed.

jindrichbenes, 2019-02-25:
Was this during printing with octoprint?

DimitarKrastev, 2019-02-26:
Yes it was

bruhmann, 2019-09-09:
I also had this happen to me today.  I'm not sure why it thought the filament ran out, as it had not run out, but the system thought it did.  I've been trying to adjust the FINDA, so maybe that improperly triggered it.  I'm also using octoprint.

JohnnyDeer, 2021-01-19:
Hi @DimitarKrastev, is this still a issue?

closed 2023-09-19 by github-actions[bot]