prusa3d/Prusa-Firmware #1572: MK3 When connecting to the printer with Octoprint the MK3 printer reboots every time.

Crunch69, 2019-02-24:
MK3 When connecting to the printer with Octoprint the MK3 printer reboots every time.

leptun, 2019-02-24:
This is normal and it is working as it is supposed to.

The reason to do so is to allow updating firmware via usb.

Is there any particular reason you would want it disabled?

mionut, 2019-02-26:
If you connect to the internal serial for raspberry pi no reboot occurs, but you cannot update the firmware over this connection. To see wich pins are for serial connection look at einsy schematic.

aemileski, 2019-03-01:
Duplicate of #1253 

See [my comment](https://github.com/prusa3d/Prusa-Firmware/issues/1253#issuecomment-439266082) for details, and hardware work-around (not recommended).

mkbel, 2019-04-09:
This is feature.
You can permanently disable this feature by command:
`;C32u2_RMD`
You can reenable this feature by command (semicolon is part of the command)
`;C32u2_RME`

You can reset the printer by command:
`;C2560_RES`

closed 2019-04-09 by mkbel

3d-gussner, 2020-01-03:
@mkbel I tried the commands via Pronterface, Octoprint even native serial communication and get always and error "unknown" command.
The `lsusb -v` command shows
For the MK3s
```
  idVendor           0x2c99
  idProduct          0x0002
  bcdDevice            1.30
  iManufacturer           1 Prusa Research (prusa3d.com)
  iProduct                2 Original Prusa i3 MK3
```
and MK2
```
  idVendor           0x2c99
  idProduct          0x0001
  bcdDevice            0.01
  iManufacturer           1 Prusa Research (prusa3d.com)
  iProduct                2 Original Prusa i3 MK2
```

mkbel, 2020-01-03:
I can imagine it is not working through Pronterface and Octoprint as this command looks like comment and those software is stripping out comments. But through Serial port terminal it works for me.

BlueHoundInc, 2020-01-23:
> You can permanently disable this feature by command:
> `;C32u2_RMD`
> You can reenable this feature by command (semicolon is part of the command)
> `;C32u2_RME`

This worked great on the MK3S on the serial terminal. I had read in some other places that the reset was needed to allow for firmware upgrades. Will I need to re-enable the reset function prior to doing a firmware upgrade or should they still work with this permanently disabled?

Thanks!

leptun, 2020-01-23:
Yes, otherwise the software that updates won’t be able to reset the firmware through the usual means.
Also, make sure to use PrusaSlicer’s firmware updater to install new firmware as the firmware updater in octoprint is not compatible (in case you want to use other languages)

easonjim, 2023-12-26:
@leptun hello leptun, Where can I find the code implementation? thanks.

jacopotediosi, 2024-03-01:
I just released a new plugin for OctoPrint that might help solve this problem. [Prusa Reset Mode](https://github.com/jacopotediosi/Octoprint-PrusaResetMode) adds a tab to Octoprint, with buttons that allow you to send serial commands directly from the OctoPrint interface to enable and disable this "automatic reset feature".