prusa3d/Prusa-Firmware #1574: Z Axis calibration fails before printing. 3.5.2 MK3

Walterenz, 2019-02-26:
I flashed to 3.5.2 1999 on my MK3. Since then i get Z axis calibration errors (without further details) at the calibration of the 9 points before starting to print. The selftest is passed without any problems. i also did the XYZ calibration. it did not help. Other users have the same trouble. Best regards, walter

mkbel, 2019-03-04:
@Walterenz Hello, can you please attach log? Connect printer via USB to PC before doing anything as opening virtual serial port resets the printer. Open virtual serial belonging to the printer in any tool allowing serial communication monitoring like Hyperterminal, Putty or Serial port terminal. Setting is Baud Rate : 115200, Parity: none, (Data)Bits: 8, Stopbits: 1, Flow cotrol: none. Log it to file and if you manage to reproduce the problem when logging, please attach that log there.

Walterenz, 2019-03-07:
hello,

i am using a MAC. I think i only have the terminal program and don´t know exactly how to start the log.
can you pls help me a little with that ;)

meanwhile, i just upgraded to firmware 3.6.0, did the whole calibration again, and at the first layer calibration (you know, when the height of the 9 points is measured) the z axis error occured again.

thanks a lot
 walter

Walterenz, 2019-03-07:
Hey….i just wrote, that i have problems doing the log, but now i made it :)

the error occoured and it is in the logfile which i attach.
it appeared at: 

echo:busy: paused for user
                                                                                                                                       echo:busy: paused for user
         echo:busy: paused for user
                                   echo:busy: paused for user
                                                             tmc2130_home_enter(axes_mask=0x04)
                                                                                               tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04

after the heating process.

i hope that helps!

> @Walterenz <https://github.com/Walterenz> Hello, can you please attach log? Connect printer via USB to PC before doing anything as opening virtual serial port resets the printer. Open virtual serial [...]

`Wecho:Now fresh file: RING40~1.GCO
                                   File opened: RING40~1.GCO Size: 5829820
                                                                          File selected
       Done printing file
                         echo:0 hours 1 minutes
                                               echo:enqueing "M23 ring40~1.gco"
                                                                               echo:enqueing "M24"
                  echo:Now fresh file: ring40~1.gco
                                                   File opened: ring40~1.gco Size: 5829820
          File selected
                       fsensor_autoload_check_stop - autoload DISABLED

                                                                      NORMAL MODE: Percent done: 0; print time remaining in mins: 95
                                                    SILENT MODE: Percent done: 255; print time remaining in mins: -1
                                    NORMAL MODE: Percent done: 0; print time remaining in mins: 95
                  SILENT MODE: Percent done: 0; print time remaining in mins: 95
T:155.78 E:0 B:46.9
                   T:155.26 E:0 B:46.8
                                      T:155.08 E:0 B:46.7
                                                         T:155.20 E:0 B:46.8
                                                                            T:155.84 E:0 B:46.6
               T:156.83 E:0 B:46.7
                                  T:158.46 E:0 B:46.6
                                                     T:160.06 E:0 B:46.8
                                                                        T:162.06 E:0 B:47.0
           T:164.89 E:0 B:47.2
                              T:167.11 E:0 B:47.2
                                                 T:169.34 E:0 B:47.6
                                                                    T:172.62 E:0 B:47.7
       T:175.16 E:0 B:48.1
                          T:177.96 E:0 B:48.2
                                             T:180.71 E:0 B:48.5
                                                                T:184.26 E:0 B:48.8
   T:186.93 E:0 B:49.2
                      T:190.13 E:0 B:49.6
                                         T:192.83 E:0 B:49.7
                                                            T:195.39 E:0 B:50.2
                                                                               T:198.45 E:0 B:50.5
                  T:200.49 E:0 B:50.9
                                     T:203.35 E:0 B:51.1
                                                        T:206.03 E:0 B:51.6
                                                                           T:208.30 E:0 B:52.0
                                                                                              T:209.78 E:0 B:52.3
                                                                                                                 T:211.51 E:0 B:52.7
                                                                                                                                    T:213.07 E:0 B:53.1
                                                                                                                                                       T:213.91 E:0 B:53.4
                  T:214.48 E:0 B:54.0
                                     T:214.90 E:0 B:54.5
                                                        T:214.95 E:0 B:54.7
                                                                           T:214.79 E:0 B:55.2
                                                                                              T:214.84 E:0 B:55.3
                                                                                                                 T:214.79 E:0 B:55.6
[... 72 lines cut ...]
                                   echo:busy: paused for user
                                                             tmc2130_home_enter(axes_mask=0x04)
                                                                                               tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04
                                                                                                                                                 echo:busy: processing
              tmc2130_home_enter(axes_mask=0x04)
                                                echo:busy: processing
                                                                     tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04
[... cut ...]
                                                                                                                         tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04
                   echo:busy: processing
                                        tmc2130_home_enter(axes_mask=0x04)
                                                                          echo:busy: processing
                                                                                               echo:busy: processing

closed 2023-09-19 by github-actions[bot]