prusa3d/Prusa-Firmware #1580: X axis auto-home failing; sensitivity too high on crash detection?

jrmiller82, 2019-02-28:
Similar to issue #1359. The auto-home feature doesn't work at all, either through the Calibrate/Auto Home on the LCD, or hitting the X/Y home in octoprint.  The x-carriage acts like it's already hit the x-stepper motor housing no matter where the x carriage is at time of auto homing. 

Machine is brand new kit mk3s. There are no obstructions, the x-axis homing works fine on the Wizard and initial self test / xyz calibration, but then auto home does not work at all afterwards. 

If I put a slight pressure on the extruder cable bundle, I can get the x-carriage to move more towards the x-stepper motor housing. 

Belts are properly tensioned, prints look great for the most part, just can't get it to auto home. 

My solution to auto homing is to turn the printer off, manually put x carriage touching to the x-stepper motor housing, and then auto-home will work fine. 

Since the x-homing works fine on the Wizard but not when I pick auto home, there's something wrong in the software of the autohome feature. I'm not alone; there are plenty of people on the forum complaining about the same problem. For example: https://shop.prusa3d.com/forum/assembly-and-first-prints-troubleshooting-f62/mk3-loses-its-home-after-stopping-a-print-t17649.html#p131138

inquisitiveKangaroo, 2019-03-01:
Hi,
thank you for the feedback. Would be possible to provide us build number of the firmware you are currently using?

jrmiller82, 2019-03-01:
3.5.3

clfaye, 2019-06-08:
I have this same problem.  Firmware 3.7.1 and 1.0.5.  The x axis thinks it's home wherever it is when i start a print.  I have to manually turn off the printer and push the extruder all the way to the left before starting a print

thegrandblooms, 2019-07-24:
I'm having this issue also. Have you been able to find a fix?

The "Wizard" x/y/z axis test has no issue, but when I interrupt a failed print or reset the machine it seems to lose the correct home position and will start to crash, particularly on the X axis. The only way to get the correct home position is to run the wizard again, which I'm getting quite tired of.

JAG836, 2019-08-05:
Same issue here. Most recently it was on the last few layers of a print, thought it crashed and stopped (I'm not sure what caused the crash, it looked ok to me). It tried to resume the print automatically but it must have thought the current position was 0,0, so it moved off into space where it thought the print was and tried to keep printing. Stops at about 14 sec into [this video](https://video.nest.com/clip/ef8750e1f6c94f9da7511494a21f6d1b.mp4) (sorry, sort of crappy quality because I had the light off).

thegrandblooms, 2019-08-05:
I've since solved this on my machine - the bearings on the X axis were fighting with each-other and increasing the friction on the X-axis. This was throwing off the auto-homing which is sensed by motor torque. If the torque is high, the homing will fail. Loosening the clamping piece on the X axis bearings could solve this for you, as they will become more aligned with less clamping force. Perhaps new bearings, if that doesn't work.

JAG836, 2019-08-12:
I believe I fixed this as well, because the printer just detected a crash but was able to auto home and resume correctly. 
I loosened up both belts a touch. I noticed if I had all 4 X-carriage-back screws snug I felt too much resistance when manually moving the extruder along the X axis (this was done after loosening the belt). Loosening the middle screw but leaving the top 2 and bottom screw snug decreased resistance and the extruder moved more smoothly. Thanks @thegrandblooms for the tip!

BasicBeluga, 2019-10-20:
@thegrandblooms this fix worked for me, loosening the x-carriage-back screws and a quick recalibration worked wonders 🙇

haybeseret, 2020-01-10:
Same here, with MK3s. adjusting the screws will not solve it for me, yet it is indeed the issue @thegrandblooms describes, where bearings on the X axis were fighting with each-other. i had to apply some lubricant to the axis (top rod) which seem to have solve it for me. It is kind of disappointing that such a well thought printer is failing on such technical manner.

thegrandblooms, 2020-01-10:
A month or two after I posted I noticed some scoring along my precision rods and the above problem returned, I've since switched to polymer bearings (and bought new rods, but never installed them) and I feel much more comfortable about the reliability of these axes, I haven't encountered the homing issue or skipped steps in a few months now. Just to give you guys an update :)

Gh0stw0lf, 2020-06-04:
Chiming into say that the loosening the screws on the x-carriage (1/8 of a turn) fixed it for me. I should have taken the hint when the motor skipped while doing the x axis indicating that the tension was WAYYY too high but hey, got it in the end.

thrasher, 2020-08-19:
+1 to "lube the rails". Unscrewing the x-carriage plate may not be enough. There was enough friction that pushing the hotend left or right would lift the printer off the table before it moved... and it turns out that's not good. Prior to lubing the rails were not visibly dirty - they were just dry, and that caused all of the friction. I used some light oil (Muck Off wet bicycle oil specifically) to lube the rails, then removed excess lube and dust with a clean towel, then again lightly re-lube the rails. I was really surprised by how much of a difference this made.  And now it's 6 hours into a big print with no problems.

Prusa-Support, 2022-06-26:
Thank you all for your feedback.

This issue will be closed because stale.
Moreover some of the participants in this conversation proved that the occurrence was not related to the firmware but a consequence of mechanical stress.

Assembly inaccuracies, wearing and lack of maintenance could be some of the causes. Also please avoid excess lubricant and super-tight belts
.
Talking about assembly, the design has slightly changed over the time to reduce chances of overpressuring and deforming the linear bearings on both the X and Y axis.

Similar problems could be resolved through a proper troubleshooting session. Please consider reaching out the Technical Support before posting new similar problems.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

closed 2022-06-26 by Prusa-Support