prusa3d/Prusa-Firmware #1587: MK3S XYZ Calibration Failure 3.6.0

jeffreyso, 2019-03-02:
Upon my build of the MK3S I built everything to spec and during the first calibration it instantly failed on the XYZ axis when it homed and went down. I tried a new PINDA, checked all axis, tensioned all belts to 260 and everything still failed. I went to forum groups and found out I'm not the only one with this issue. Its a firmware issue as I downgraded to the MK3 firmware and was able to get the calibration to work. Here is some reasons that people have suggested. Please fix this and release a new firmware ASAP as I am quite disappointed after purchasing two MK3S's and they have this issue. If however I issue a G80 command to the printer it will fail on the 3rd calibration point and completely hit the bed. 

- It fails on the 3rd or 4th point. There are two linked issues, a design issue in the bed and the XYZ calibration algorithm that is not good.

- XYZ algo is first doing circles to detect Z level. This level is then used to find the pinda circle. The problem is that on back of the bed (point 3 and 4) you have a big surface of copper and pinda might see it "too quickly" compared to smaller thinner copper traces around. So it gets a bad level and cannot find pinda point.

mkbel, 2019-03-04:
@jeffreyso Hello, can you please attach log? Connect printer via USB to PC before doing anything as opening virtual serial port resets the printer. Open virtual serial belonging to the printer in any tool allowing serial communication monitoring like Hyperterminal, Putty or Serial port terminal. Setting is Baud Rate : 115200, Parity: none, (Data)Bits: 8, Stopbits: 1, Flow cotrol: none. Log it to file and if you manage to reproduce the problem when logging, please attach that log there.

mkbel, 2019-03-04:
@jeffreyso As a workaround you can use latest firmware without XY calibration. XY calibration doesn't have much sense for MK3S printer, as this is for compensating non-perpendicular X and Y axis, it was possible to assemble MK2 printer without having X and Y axis perpendicular, but MK3 aluminum extrusion frame is always perpendicular. I don't know, why we still didn't remove this calibration for MK3/MK3S printers. Do factory reset and just run selftest, Z calibration and first layer calibration.

jeffreyso, 2019-03-04:
My problem is that I can’t even do a g80 mesh bed level. It crashes on the 3rd point.

mkbel, 2019-03-04:
Ok, your second issue is already reported https://github.com/prusa3d/Prusa-Firmware/issues/1574, you can watch it or contribute log there.

PavelSindler, 2019-03-05:
Can you run Z-calibration (menu Calibration -> Calibrate Z) with steel sheet on? Mesh bed leveling may fail, because you have outdated (MK3 geometry) Z-leveling calibration data.
If you do Z-calibration, you should pass mesh bed leveling then.

jeffreyso, 2019-03-05:
What changed in the geometry? I’m using custom parts so I’m wondering if that makes a difference as the PINDA position hasn’t moved so it shouldn’t recognize that the extruder is still in the same position as the MK3. Did the carriage change/did it move closer back? I’m using vekoj tensioner x axis with the R3 carriage and a bondtech BMG extruder. Please let me know. The calibration fails upon the XYZ calibration right when it homes and the nozzle comes down to the first point.

henrikkoch, 2019-03-19:
After upgrading parts from Mk3 to Mk3s and flashing with new beta 3.7.0. RC1 I now can't do XYZ calibration. PINDA is fully functional in all places and can both trigger LED and menu. So everything seems connected. But the Printer is not able to calibrate anymore. Says "XYZ Calibration failed. Bed calibration point was not found....."

mkbel, 2019-03-20:
@henrikkoch  Hello, can you please attach log? Connect printer via USB to PC before doing anything as opening virtual serial port resets the printer. Open virtual serial belonging to the printer in any tool allowing serial communication monitoring like Hyperterminal, Putty or Serial port terminal. Setting is Baud Rate : 115200, Parity: none, (Data)Bits: 8, Stopbits: 1, Flow cotrol: none. Log it to file and if you manage to reproduce the problem when logging, please attach that log there.

henrikkoch, 2019-03-21:
@mkbel :   I sorted it out. It was a false conclusion and it was due to the X-belt was ONE tooth too tight and then the printer was having trouble to move the extruder easy enough. Before coming to the final conclusion I also tried with the previous release version of firmware. That had the same problem. I then relaxed the belt ONE tooth and both version were then working OK afterwards

mkbel, 2019-03-21:
I am closing this, as @jeffreyso is unable to produce log and no one else is affected.

closed 2019-03-21 by mkbel

RawSmokeTerribilus, 2019-07-16:
Hi everyone! i had the same issue after a full bear upgrade. The printer was homing almost perfectly (the PINDA probe stops right on the left line of the circle), my belts values on firmware was 248 and 260, so i thought that wasn't the problem... As i've checked everithing twice without find a reason to the xyz calibration error (on the first point, with the message "calibration point was not found"), what i did was to flash the firmware from 3.5.2. to 3.7.2, go further the Z calibration with the pei sheet, then the mesh calibration, then the PID calibration, then the temperature calibration and now everything seems to be working, i did too a first layer calibration and it's fine. On the lcd is telling me that the mk3 is ok... Even if i never did the XY calibration.
So, i have two questions... Is my routine dangerous for the printer? Do you still needing a log? if you want a log, i can do that for you, no problem.

macraj, 2019-09-24:
Hi! I have the same issue. I am crazy about this it's software related. I can't calibrate Z-axis and I can bet that most of Prusa users have the same problem but they don't know it yet.
And if I remember correctly, the problem started after upgrade to MK3S.

macraj, 2019-09-24:
When I downgraded software to 3.1.2 calibration Z-axis works like charm.

leptun, 2019-09-25:
Wait that is dangerous. You are using a firmware that was made only for the MK3. You will have issues with the filament sensor and you expose yourself to older bugs.

macraj, 2019-09-25:
I know this, thx for caring. I had to check if it works (is this software related or hardware) or not. Of course works, but filament sensor is on different port and not working. I think Prusa is like ostrich. He puts his head in the sand. After all, it is a commercial product and and should support their clients. My confidence in MK3 fell, by the way I bought Creality CR-10s pro and I'm very impressed with this printer (after few little mods). Now I will try Klipper software I heard its pretty good.

kellycoe, 2020-10-26:
I just went through a 4th position xyz failure about 6 times.. adjusting the PINDA and going through the wiring to see if it was hanging low (discovered a zip tie that touched the cpu box was enough to mess this up) 
and only when I was trying for the 7th time did I notice that the shaft on the x motor would slip and the gear would stop spinning momentarily... so.. note to self..  check all the belt drives..  
with that tight.. 8th time worked perfectly..
nice!!!   thought I'd share

macraj, 2020-10-27:
For me, it was a problem with custom (self-made) bed supports. They were a little taller and software have fixed Z-axis length so it didn't work properly.

On 26 Oct 2020, 22:58 +0100, prusa3d/Prusa-Firmware <reply@reply.github.com>, wrote:
>
> thought I'd share