prusa3d/Prusa-Firmware #1603: Invert logical IR sensor MK3S

GastonGelhorn, 2019-03-06:
Hello,

How the logic of the IR sensor could be inverted in the firmware input.

Thanks

3d-gussner, 2019-03-08:
@GastonGelhorn At this moment there is no invert definition like for the endstops `X_MIN_ENDSTOP_INVERTING`
But if you search for `IR_SENSOR` in the code you can find it and adjust it to your needs.

I guess you are trying to use another switch.

Hope that helps.

GastonGelhorn, 2019-03-08:
Exactly is another non-original switch, but it is NC and it shows 1 when it is 0 for no filament in the extruder.
I try to change sometime in the IR_SENSOR 0 by 1 and 1 by 0 true for false, etc., but I do not get the changes or the expected result.
I will try to use another NO switch, so as not to change the firmware too much.

Many thanks!

3d-gussner, 2019-06-16:
@GastonGelhorn Did you find another switch? If so please close the issue

GastonGelhorn, 2019-06-16:
No, Ill modify the 3d part for invert function

closed 2019-06-16 by GastonGelhorn

imaluschi, 2019-09-17:
For my MK2 following solution worked for inverting the filament runout-sensor.

Taken from (thanks to Stephan):
https://drucktipps3d.de/filamentsentinel-von-dyzedesign-am-prusa-mk2s/

Change marlin_main.cpp, row 2088
if(!READ(FR_SENS)){
instead of
if(READ(FR_SENS)){