prusa3d/Prusa-Firmware #1606: Bed Preheat Error 3.6 mk3

nato2k, 2019-03-08:
With 3.6 I am getting a bed preheat error when printing ABS in the middle of the bed.  It can be fixed by moving the print to the front or the back of the bed.  I ended up going back to 3.4.2 which does not have any issue.

Basically when the bed is going from 100 to 110, at some point during the heat up I get the error, it appears that it is taking longer than expected.

zhangsiyan12134, 2019-03-12:
Yeah, I'm having exact issue with my 2 month old MK3 on 3.6 firmware. The problem happened at the exact place after 3rd layer while heated bed temperature need to go up to 110. I'm using the default HIPS setting with 0.2 mm layer height.

WIGG77, 2019-03-17:
> Yeah, I'm having exact issue with my 2 month old MK3 on 3.6 firmware. The problem happened at the exact place after 3rd layer while heated bed temperature need to go up to 110. I'm using the default HIPS setting with 0.2 mm layer height.

Same here. ABS fails on 4th layer. 
Firmware v3.4.2 does not have this issue so it's defiantly not the printer. Unfortunatly though I was getting an annoying beep when you turn the printer on and also upon print start with a message that new firmware was available. I've since gone back to 3.6 but not used ABS for a few weeks because of this problem.

aeresov, 2019-03-18:
alright on 3.6.0, but started getting same on 3.7.0-RC1

awesinine, 2019-04-15:
also just ran into this error.  printing abs, small part in the center, printer seems to have trouble reaching getting from 100-110 and then throws the error.  3.7.0

Prusa-Support, 2022-05-13:
Hello! I am sorry for the very late response to this thread. Is the issue still relevant with the newer firmware versions released in the meantime? In case this issue is still there we can look into it. 

Alessandro Pantaleo
Prusa Research

closed 2023-09-21 by github-actions[bot]