prusa3d/Prusa-Firmware #1609: Firmware 3.6.0 left hotend fan fault.

3D-Q, 2019-03-09:
I just upgraded my I3 MK3 firmware 3.5.1 to 3.6.0 and now i get a fault when running the wizard, the (left hotend fan dont work).
any ideas as why?
When downgrading to version 3.5.1 it work all fine?
The printer is from december 2018

Panayiotis-git, 2019-03-09:
And you answer that is works?

F0x06, 2019-03-09:
Probably related to https://github.com/prusa3d/Prusa-Firmware/issues/1520, normaly FW asks for manual verification, this should work

3D-Q, 2019-03-10:
3.5.1 works no problem.

Prusa-Support, 2022-06-02:
Hello! Sorry for intervening so late on this. Was this fixed with the successive firmware versions released in the meantime? Or was it related with selecting "yes/no" on the manual fan check when the printer asked if the fan is spinning? 

I'll close the issue for now, but in case of more aspects to further review, do not hesitate to let me know info@prusa3d.com mentioning me, including a video of the failed self test to better review. 

Alessandro Pantaleo
Prusa Research

closed 2022-06-02 by Prusa-Support