prusa3d/Prusa-Firmware #1610: feature request: print autostart on specific filename

m-cas, 2019-03-09:
I'd like to have the possibility to put a file with a specific filename, e.g. __AUTOPRINT.gcode, and have the printer start this automagically.

cimoalpacino, 2019-03-10:
I think this is already implemented.
You have to name your file as auto0.g to have it automatically run after boot up sequence.
If you want another file to start after this, you name it auto1.g and so on. 
Files must not be placed in subfolders.

m-cas, 2019-03-10:
Cool! You mention boot up, will it also work on every card insertion (printer already booted)?
-
From my phone while mobile, please excuse any errors.

cimoalpacino, 2019-03-10:
Don't know. I don't use this feature. 
Try it out and let us know. 
Cheers

cimoalpacino, 2019-03-10:
Also, if it works, it would be nice to close this thread 😉

m-cas, 2019-03-10:
tested it, it doesn’t work on card insertion. Request remains then with addendum “… it should also work on card insertion"

michalxfanta, 2019-10-11:
Thank you for your interesting idea, we will add it to our backlog.

closed 2019-10-11 by michalxfanta