prusa3d/Prusa-Firmware #1611: Building 3.6.0 and later; tons of undefined references to millis2, etc.

Murloc992, 2019-03-09:
I have previously built the 3.5.0 firmware no problem. Updated to fresh 3.6.0 via github, made my changes and tried to build. H E C K O F A L O T of: `ccW8uvRV.ltrans25.o*: (.text+0x78a): undefined reference to millis2`.

I have changed LANG_MODE to 0 as before.

The only change I did personally was to go to PT1000. That's all.

Any insights?

Murloc992, 2019-03-09:
It fixed itself after checking out and building for the 10th time. Interesting.

closed 2019-03-09 by Murloc992

maximlevitsky, 2019-03-10:
@Murloc992 This is off-topic and on-topic at the same time:
https://www.monkeyuser.com/2018/define-madness/

My guess is partial build using out of sync object files, and this was fixed by cleaning and building again.