prusa3d/Prusa-Firmware #1613: Self test error

fucosk, 2019-03-10:
Since version 3.5.2 (3.5.1 was OK), on MK3 the callibration wizzard and/or self test fails on "left extruder fan connection error"). The error is probably in this commit: https://github.com/prusa3d/Prusa-Firmware/commit/f9298b37b802832705009d3c73dbf2994f9a234c#diff-0e72dfbd236a78609a669f8dc5db0f5f There is an forum thread: https://shop.prusa3d.com/forum/assembly-and-first-prints-troubleshooting-f62/left-hotend-fan-spinning-not-spinning-mk3-t28000.html As well as test case: https://www.youtube.com/watch?v=lnyx6YotFUU

Panayiotis-git, 2019-03-10:
I do not understand why PR enabled these questions for the MK3, while it can automatically detect if they are spinning.
But if the user replies with the correct answer when the printer asks (if the fan is spinning at the moment of the the question) then the self test proceeds.

fucosk, 2019-03-10:
I only guess it was meant as a swapped cables test.

cobramike, 2019-04-22:
After updateing the prusa firmware. I get the same error Front print fan check wire. 

But when i do the test. 
Extruder fan 👍 
Front fan fails. Its testing for 15 sec and then the question is comming. When pres ok its saying front fan error check wire.

When wil prusa fix this.

fucosk, 2019-04-23:
> When wil prusa fix this.

I do not think they are going to fix it, as they do not understand the problem. In fact, it is not an error. I was also fooled by the creepy user interface/experience. The so called "self-test" is no longer self test. It is an interactive test. You have to respond the fans state also in case everything is OK. The stupid thing is, you see one fan is OK, and then, when self-testing part assumes the other one is also OK, it won't display it is OK, instead it is going to ask you the question about the left fan. And it asks you, if the left fan is spinning in the time it is actually slowing down to zero. You should answer it is not spinning. And then, there is asking you about the second fan but you can easily overlook it and think it is asking you the same question again. So I thought it is an deadlock. When you read carefully and answer slowly (waiting the fan to astually not spin), you will get to the successful test. Then the UI displais you also the second fan is OK. What a dumb person could design this... :-) See my video https://www.youtube.com/watch?v=lnyx6YotFUU and the prusa forum thread: https://shop.prusa3d.com/forum/assembly-and-first-prints-troubleshooting-f62/left-hotend-fan-spinning-not-spinning-mk3-t28000.html (now down)

wavexx, 2022-04-04:
Found this while searching for old issues. PR #3461 should improve the behavior of this check.

closed 2023-09-19 by github-actions[bot]