prusa3d/Prusa-Firmware #1617: Nonsensical PINDA / Bed Levelling Values

Enverex, 2019-03-10:
I had a print suddenly decide that (on the first layer) it should instead shoot nearly 10cm in the air and carry on printing there.

[image]

I assumed it was a slicer error, so I resliced, saved and reprinted. I forgot that I'd stored the PINDA reading though, and this is what it showed:

[image]

A few things worry me here. First is that it freaked out in the first place and the second is the lack of sanity checking. I wouldn't expect the firmware to accept bed variation levels of nearly 10cm.

Enverex, 2019-03-13:
After running the test again, it's still reporting completely silly values for one of the interpolated points. 3cm this time, same area though:

[image]

Enverex, 2019-03-13:
It does normally work though, for example here's what it looks like typically. 
[image]

inquisitiveKangaroo, 2019-03-14:
Hi,
thank you for your feedback. We are already working on optimalization of MBL so we will prevent such behaviour from happening.
Have a nice printing day.)

leptun, 2023-05-13:
There have been safety checks in the MK3 branch of the firmware for a long time that should prevent something like this from happening. So the MK2.5(S) and MK3(S) are safe from this.

closed 2023-05-13 by leptun