prusa3d/Prusa-Firmware #1618: confirming that 1st layer calibration will use 7x7 leveling

EXOgreen, 2019-03-10:
I'd like to confirm that with the next update the 1st layer calibration will also use 7x7 mesh bed leveling instead of 3x3. When using @mionut 's firmware that I compiled this is not how it works, even though this would be the best for that first layer calibration.

jhoblitt, 2019-03-12:
#1586 does not change the default value of `MESH_MEAS_NUM_X_POINTS`.  This means that the first layer calibration wizard still does 3x3, which is then interpolated into a 7x7 matrix.  Assuming a `G80 N7` is done as part of every print this shouldn't really matter and 7x7, but I agree that default should probably be changed.

michalxfanta, 2019-04-09:
Since the FW version 3.7.0 is a 7x7 mesh bed leveling implemented.

closed 2019-04-09 by michalxfanta