prusa3d/Prusa-Firmware #1621: MK2.5S 3.6.0 firmware Issues

tonysmartantzoulatos, 2019-03-13:
Hello Firmware team,
I hope this message is placed on the proper place for you guys to see.
Yesterday i made the following post on the Prusa Community FB Group and since github is the place for firmware i decided to raise it to issue.
OK, seems i am out of ideas and need the mindhives brainstorming:
On our MK2.5 we had baught an MMU2 which was only slightly tested and currently not running, which means that we qualified for an S upgrade that was received some time last week. This printer runs on a full bear frame that was running flawlesly with a filament sensor and when the 3.5.1 from Stephen Armstrong was uploaded it was even better.
Since we received the S upgrade, decided yesterday to install it and in the process, clean Z and X bearing, and check all plug and wires on the rambo.
Since it was all taken apart i decide to run the wizard that constantly failed on EX-Heater/thermistor.-> Stop it and run a preheat.....guess what, runs fine!. -> Check thermistor, try a few times again, fail, fail fail. Change Thermistor, fail once more ->Contact support, i am given advice (that made no sence) to factory reset and burn firmware again, i do that and guess? Same issue!, decide to ditch wizard and go with XYZ, WHAT THE...the whole thing its been working with no hickups for a long long time and each month it receives its love and care with occasional changes on nozzle, thermistors or what might be needed. On the MK52 bed calibration 1/4 fails to be found 3 times, on the 4th try fails to find second and on the 5th finds all. (YES PROBE IS OK since after this, Z height was tuned to -0.390) -> test a couple of more preheats and on PET (right before it had done ABS) i get BED PREHEAT FAILED and on second attempt i get with 40+c on the bed, a BED MINITEMP ERROR. -> I do the 3.6.0. for the MK2.5S firmware once more run all once more (no wizzard, but XYZ) runs ok, got once the preheat error and at the moment its printing.

Any ideas?

Since yesterday i have the following results:
WITH more than 15hours of prints on this S upgrade printer since the "rebuild", Still fails to pass hotend test on wizard (while hotend measure and functions properly), still gives bed minitemp error on strangest occasions and also has given me twice more on preheat BED PREHEAT FAILED.
All in all this has nothing to do with my rebuild but rather the firmware it self. I found that even though Slic3rPE gives Firmware upload success, from the advanced tab i can see an AVRdude error at the begining of the update. I can provide any additional information required, but please dont ask me to go on online support, since no matter how much those people want to help, plenty of them dont have more than preset responces to post questions.

tonysmartantzoulatos, 2019-03-14:
In addition to the problems stated above i have found also that even though my Z height is -0.135 for a perfect first layer in most cases while on G80 it stops and says Z enforcer. Asks i do the Z calibrate and then starts ok.
For those which will say i must check my build, this frame is 100 correct and with the 3.5.1 FW was printing flawlesly. only the extruder was taken apart.

RosieTheRabbit, 2019-03-18:
I am not usually one to say - "ME TOO" but "me too".  I recently did the upgrade from mk2s to mk2.5s.  Was running OK until recently.  I now get frequent "Some error has occurred Z axis recal enforced" and can barely get a print going without the "BED PREHEAT FAILED".  So far Prusa support via their text support have been of no help.

Sorry to do this but I am getting frustrated with the software / firmware issues messing me up.

Cheers

tonysmartantzoulatos, 2019-03-19:
Few days ago i posted this on the facebook group:
MK2.5S Users update:
The 3.6.0. firmware is for sure full of bugs, since i just rolled back to the 7x7 3.5.1 i was using and printer does not show all the "issues" on wizard that did with 3.6.0 but yes it fails on sensing the new filament sensor. Preheat also works without any problem. Last but not least on the std X axis STL the new R4 endstop is wrong by almost 3mm, since i could notice homing on the 1st probe point by almost 3mm off. I cut it down and instalty home went on 1 point center (maybe a few tenths off but i will investigate more).
If allowed by the moderators of this page i will post the new endstop stl for you guys to test.

Today i started having non stop BED PREHEAT FAILED AGAIN on the second layer where i asked the temp to jump from 80c to 90c. i just run a PID calibration on the BED but the problem persists. Seems as if with the 2 new firmwares the bed temperature ramps up very slowly from 80-90C and an error kicks in. Just seems strange that this is the error we get and not HEATING FAILED, or what ever it was called in the past.

tonysmartantzoulatos, 2019-03-19:
As soon as i did the previous post i decided to slice the same print with only 1 difference in settings.
Instead of 80c first layer on the bed and 90c on seccond i tuned to be 85c on all.
On all previous attempts the error would happen when the printer was trying to just from 80-90 and it was doing so in a really slow fashion. 
Now this print has been into the 5 layer and still printing.
There seems to be a bug and the team must check it.

tonysmartantzoulatos, 2019-04-02:
Even though it feels as if the developers have not even bothered to read this "issues" post i will continue to update.
This morning i updated my printer with the new 3.7.0"stable. Firmware update said success, even though i could find on the advanced tab an AVRdude error in there. I tried it a couple of times and the same error came up.....and yes i used the 1.3rambo FW, since both my printers have 1.3rambo on them and i have been updating them for some time.
All the bed related temperature errors are still here!!!
I also tested as per online support to unplug the new S, ball actuated filament sensor only to find that after that the printer would heat up perfect or preheat perfect, but as soon as 7x7 probing was done, instead of a purge line it would jump to a change filament routine. I would take filament out, put back in, probe once more and...................jump to a filament change routine........and again and again and again, until i would power it off, plug the sensor, get the bed temp errors and as soon as i was able to heat the bed to desired temp it would start printing fine.
I hope developers spend a couple of minutes to read this post

michalxfanta, 2019-04-02:
Hello, could you send to us more info about AVRdude error and the log from the flashing process? And also send to us GCODE with bed temperature changing during print.

tonysmartantzoulatos, 2019-04-02:
Here is the file and i will try in a while to upgrade firmware once more and printscreen the error.
[antenna 1_0.2mm_PET_MK2.5S.zip](https://github.com/prusa3d/Prusa-Firmware/files/3033704/antenna.1_0.2mm_PET_MK2.5S.zip)

tonysmartantzoulatos, 2019-04-02:
at the moment i cant connect my PC to the printer since its working on other projects. you will have a picture here in a few hours

tonysmartantzoulatos, 2019-04-03:
@michalxfanta 
I just run the firmware again and here are the pictures you requested.

[image]
[image]

tonysmartantzoulatos, 2019-04-04:
@michalxfanta, i read this morning on the FB community forum page that you guys are not able to reproduce this problem on your machines, thus this is why after all this time it has not been solved.
Since apparently its being produced on many machines worldwide and in my personal opinion its related to the filament sensor (or some part of the firmware that its being confused by it), Would there be a series of tests that you would like me to do and video at the same time in order for you guys to see it live?

tonysmartantzoulatos, 2019-04-10:
@michalxfanta as per @Nippylicious on another preheat issue post i can confirm his findings that if you preheat to (in my case) 29c and then preheat to target temp you dont get a preheat error, while at the same time it remains that if you set 80c as first layer bed temp and increase this temp on second layer even by 5c, the you will get a bed heating error, while on the same time if you make 120c or any other bed temp on all layers, you dont get any bed heating error.

tonysmartantzoulatos, 2019-04-27:
@michalxfanta Another piece of information that i realized today is that this morning even though my printer was below 20c (and the first preheat error comes around 29-30C on the bed) i pressed unload filament on PETG profile and the printer heated up to unload temps with no issue. The bed had only reached 65c by then, but still there was no 29-30 first error and a need of reset.
Has there been any update on this issue?

clintonefisher, 2019-04-29:
3.7.0 and I get the MINTEMP BED error when preheating from cold and it hits about 27C It's not always the same temperature. 

Printer is:  MK2.5S+MMU2S (MK2S --> MK2.5 --> MK2.5S+MMU2S)

I wonder how many others are affected and I wonder if not that many people, then could it be some leftover EEPROM values or something from an older firmware?
With the MMU2S, I upgraded my firmware to 3.6 from an older 3.1 or 3.2 version, which threw the MINTEMP BED error, and then 3.7 threw the error as well.

Don't know about others, but seems to me to be a moderate-to-high-severity issue if a lot of folks are affected.

tonysmartantzoulatos, 2019-04-30:
@michalxfanta 
After alot of testing and investigation i think i was able to track down the source of the problem for this "firmware" issue. Since i made a related post on the facebook group yesterday https://www.facebook.com/groups/prusacommunity/permalink/987465131594219/
I would suggest investigating the actuall pins for BED power on the Rambo/Einsy boards.
It seems that even though the bed plug as also the pins from first look when investigating can seem perfect, after removing the pin mount from the board in order to be able to investigate the pins properly, it is visible (as also seen on one of my pictures on that link) that due to most likely vibrations on the printer, there are signs on arcing decay on the pins, and i fact on the specific board (that when i installed 3.7.0, as a test and moved it on my 2.5S cause also "thermal/heating" issues), the actuall plugs of the extruder and the bed look to be in perfect shape.
In conclusion and after a month of investigation and testing i can 99% say that no matter if the wiring/plugs/board connectors, seem fine on the outside, you must take them apart to see this happening and i can say that these plugs are not safe for such vibrations abuse on printers. I my case and now i am on the third day of testing on the 2.5S i installed bullet plugs on the bed and extruder, tested both 3.6 and 3.7 and so far all the heat issues that were there for a month are gone.
P.S.: when i started having the "heating" issues, all plugs seemed to be in perfect shape, since they are being investigated monthly as a precausion. 
Review the picture in detail that shows the pins decay.

RosieTheRabbit, 2019-04-30:
I concur.

I had the same issue with bed preheat errors.  After a lot of testing I noticed that the bed was slow to heat.  I checked the connector on the RAMBO and found it was a bit cooked.  Rather than replace the connector with the same type, which has been known to cause problems occasionally, I used a different method.  It's a lot more DIY but I used screw connector block (the type used in home electrical installations) and cut off the plastic.  I then connected the wiring directly to the block and screwed it down tight then placed the block over the connector pins (after I'd cleared the somewhat molten plastic) and screwed that down tight.  The result is a perfectly good connection that can be checked for tightness if needed.  Since doing this I have not had the problem.

Hope this helps.

tonysmartantzoulatos, 2019-04-30:
@RosieTheRabbit 
So far all preheat issues have gone away any most bed tempo changes between 1st and 2 layer if you temps are below 100c. If you try to go above 100 and difference between 10-15c from first to second, 99% of the times you will get an error. if you keep the change to 5c difference you are 50-50. This seems to be the FW that persists

closed 2023-09-19 by github-actions[bot]