prusa3d/Prusa-Firmware #1622: Capacitor explodes on Einsy Rambo 12v bed

leptun, 2019-03-13:
I recently modded my Prusa Mk2.5 with an Einsy Rambo board and other required components (new print fan and hotend heater).

To be able to use the old bed and power supply I added a boost converter so that I get 24v for the tmc2130 drivers. The hotend also ran at 24v, but the bed input on the board was connected directly to the power supply. Since Einsy Rambo has separate power rails this should have worked.

After assembling everything I printed some parts and it worked perfectly. The firmware was the same as for Mk3, but with power panic disabled (recompiled from latest source). It worked for the first two parts, but on the third one after a few layers it decided to make a lot of smoke from one of the capacitors.
[image]
In the picture you can see on the left the capacitor that blew and the mess it has caused. Interestingly the board was still functioning and I reset it for the print to stop. I took out the board, inspected it and also cleaned it. I put it back in and found out everything worked, except the fact that it was blazing hot.
[image]

I added a fan to blow on that area since i was unable to do anything about that capacitor at that moment.

Later I bought a new capacitor with the same capacitance, but a grater rated voltage (shouldn't be a problem). I connected it along with the wires since after looking in the board schematic concluded that I could connect it between pins 4(-Bed IN) and 5(+Bed OUT) of the screw terminals, "replacing" C27. I preheated and concluded that it also began heating as soon as it started doing fast PWM on the bed. I disabled SYSTEM_TIMER_2 so that it would do a slower PWM and the capacitor stays  cool even with the bed at target temperature. The commit that changed this is 273554. Does the input voltage have a change so dramatic that it causes the capacitor to explode considering that it is a lower voltage and not using it out of spec.

I think this is a serious issue and needs to be checked so that people don't burn their boards.

If you need any more info, don't hesitate to ask me.
The firmware I used is found at https://github.com/leptun/Prusa-Firmware/tree/MKB. You can see the changes I made by looking at the commits.

Is it possible to use a lower PWM frequency on the bed but still use SYSTEM_TIMER_2?

F0x06, 2019-03-17:
Thanks a lot for pointing this, i also noticed high ambient temps on my board, and suspected the recent High PWM changes on heatbed to be the cause, i patched my firmware to use slower PWM like yours `5` instead of `1`, way cooler now !

leptun, 2019-03-17:
Update:

I tried using different prescalers on timer0 to try to improve board temperature. I did a change (not public as of yet) that allowed me to use the timer in between prescalers by limiting its TOP value and dividing OCCR0B value with a sub-prescaler. I lowered the frequency to the point I started hearing it and it was very annoying. Unfortunately the board was still heating even when I started hearing the high frequency.

After this I tried lowering the frequency and it helped a lot (capacitor cool to the touch), but the clicking was more annoying than before using timer2. It was bearable though, almost being inaudible when printing. That was the minimum frequency able to be done by that timer without software aid. Probably that using an ISR to do a software timer might also work, but then why would you use SYSTEM_TIMER_2 at all then.

Hope they can find a way to make the boards cool, while still using a high frequency.

Panayiotis-git, 2019-03-17:
@leptun I've also tried the change to 5 and it worked. I can touch the MOSFET and it is just warm. With the original firmware, it is too hot to touch.
Not: I've never had the clicking problem.

mionut, 2019-03-17:
The ambient temperature measured on the board (A: value in octoprint) for bed heated to 90 deg C with latest firmware is 61 deg C vs 44 deg C with timer2 prescaled by 1024 (with `TCCR0B = (5 << CS00)` instead of `TCCR0B = (1 << CS00)`), so there is a 17 degrees increase in temperature.
The audible noise with prescaler is not so disturbing for me.

F0x06, 2019-03-18:
@mionut For the noises, it's acceptable for me also with `TCCR0B = (5 << CS00)` way less noisy than older firmwares :)

leptun, 2019-03-18:
Well at least for you it works perfectly. I think the noise problem with mine might be the THT capacitor I added to the screw terminals? I actually have no Idea why it would make a noise difference. But considering there still is a blown capacitor on the board, that sure isn't helping...

Panayiotis-git, 2019-03-18:
Perhaps you should remove the blown capacitor. It could affect negative the circuit.

leptun, 2019-03-18:
I am worried not to break anything since the area is very packed. I might be able to pull it off with pliers, but am afraid of breaking it off with some of the board traces.

Panayiotis-git, 2019-03-18:
@leptun, your problem is not the voltage itself. It is the current. The 12V MK52 boards draw twice the current of their 24V siblings. This increases exponential the MOSFET thermal losses.

leptun, 2019-03-18:
Are you sure? That would only happen if you ran the 12v bed from 24v, but the bed is receiving 12v directly from the power supply. Would that cause more heating even though it is ran at the right voltage?

I thought the bed was using less than 15A, if it exceeded it the fuse would blow. I think the bed is ran exactly as on MK2.5 with miniRambo.

leptun, 2019-03-18:
> Perhaps you should remove the blown capacitor. It could affect negative the circuit.

I intend to buy soon a replacement capacitor and desolder that one, clean the board even better and solder the new one.
I see two options right now:
- Using the same capacitor as on the Einsy board which would probably make it heat up just like the others. [link](https://ro.mouser.com/ProductDetail/Nichicon/UWT1V101MCS1GS?qs=sGAEpiMZZMtZ1n0r9vR22cpFQt4dfSbql65FWznHrgYAtnLg%2F2WPFg%3D%3D)
- Use [this capacitor](https://ro.mouser.com/ProductDetail/United-Chemi-Con/EMZR500ARA101MF80G?qs=sGAEpiMZZMtZ1n0r9vR22cpFQt4dfSbqD9LHuQFqP18zkYrNEgoO7Q%3D%3D) which has the same ratings regarding capacitance, higher voltage tolerance and the best of all: much better ripple current filtering. It is a low ESR variant.

What do you think about this? Which one should I choose?

Panayiotis-git, 2019-03-18:
The 24V MK52 heated bed is 200W (23.3V x 8.4A). If the 12V MK52 bed is also 200W then the current through the MOSFET is 16A. I am sure that at your case the MOSFET temperature must have gone wild. The einsy protects the bed circuit with a 15A fuse.

I suggest to use an external MOSFET.

leptun, 2019-03-18:
Hmmm... I was unable to find the bed power of the MK52 12V bed. It probably isn't 200W since the MiniRambo also is only rated for 15A and there is a fuse in the way, so it should have blown. I will check the current with an ammeter if I can find one that can handle 15A.

Does anyone know the power of the 12V variant MK52 bed?
I would have used an external MOSFET if it weren't for the fact that with slow pwm the board and cap are both cool. and everything is working as supposed to.

Anyway, I think I'll be closing this issue soon since I don't think this is a place for hardware problems.
I still think it is a serious issue regarding heating MK3 boards, but there is already another open issue for that. At least that is only a software issue.

Thanks for everyone's help. You're the best.

closed 2019-03-21 by leptun