prusa3d/Prusa-Firmware #1628: feature request: set z-adjust after filament change

m-cas, 2019-03-14:
I use different z values for my filaments/profile combinations. It would be nice to be able to change or confirm the set z-correction value after the filament loads so that it will be (mostly) not necessary to change it when starting the first print with the new filament (and that I pretty much always forget) :D

Harpalyke, 2019-03-18:
I've also been thinking recently that multiple z-offset profiles could be a useful feature. 
When I print PET-G I use the smooth prusa sheet but when I print PLA I find that the buildtak flexplate works a bit better. Switching the live-z value when changing plates can be a bit tedious.

Maybe even using an NFC sticker to each sheet to automatically select the z profile based on the sheet. Would need to add support for a sensor under the heated bed for this though.

m-cas, 2019-03-18:
I guess it is a question of setting priority to the use cases and ease of implementation:
- prio1 (assume that most people use one bed and only switch material)
  add the call to the Z-setting routing when changing filament so that basically all changes needed are done. All needed code is already in place.
- prio2 (requires bigger code changes)
  move the problem completely to slic3r and allow for a matrix of different z-values combinations (material, layer thickness, bed) and have them sent over to the printer via g-codes. This will need to be mostly written anew in slic3r and would remove the need to introduce HW changes/complexity (and costs) like NFC on the bed. On the printer side you would only need to ask for confirmation if the right material/bed is loaded.

michalxfanta, 2019-04-09:
We are planning this feature in next releases.

m-cas, 2019-04-09:
cool, thanks a lot!

michalxfanta, 2019-10-11:
This feature is implemented in 3.8.0 firmware.

closed 2019-10-11 by michalxfanta