prusa3d/Prusa-Firmware #1630: Can't connect to Octoprint after mk2s>mk2.5s upgrade

ogland, 2019-03-14:
So I finished the mk2.5s upgrade from mk2s earlier this evening. I have been running dual octoprint instances from a RPi 3B+ which has been working great.

Once upgraded and updated with firmware 3.6.0 for mk2.5s I can't connect the printer to Octoprint anymore. It has a connection and the printer restarts at connection attempts but it always times out.

I have tried:
Fresh octoprint install 
Turning filament sensor on and off
Firmwares 3.1.4, 3.4.0 and 3.6.0
Powered usb hub
Baud rate auto and different manual settings

And a plethora of different stuff. I'm at a loss and I really need this to work.

ogland, 2019-03-15:
If I downgrade to mk2s firmware it connects. This isn't really a solution though.

leptun, 2019-03-15:
Do you have the filament sensor enabled? I also have this problem and it seems to be fixed by enabling the filament sensor.

ogland, 2019-03-15:
> Do you have the filament sensor enabled? I also have this problem and it seems to be fixed by enabling the filament sensor.

> I have tried:
> Turning filament sensor on and off

leptun, 2019-03-15:
Do you actually receive data from the printer when connecting? Look in the terminal for “start” from printer.

ogland, 2019-03-15:
> Do you actually receive data from the printer when connecting? Look in the terminal for “start” from printer.

Yes. I am able to update firmware so a connection is there. The "start" seems to be in a conditional in the firmware of mk2.5s/mk3s which it isn't in the mk2s.

```
	MYSERIAL.begin(BAUDRATE);
	SERIAL_PROTOCOLLNPGM("start");
	SERIAL_ECHO_START;
```

```
	MYSERIAL.begin(BAUDRATE);
	fdev_setup_stream(uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE); //setup uart out stream
#ifndef W25X20CL
	SERIAL_PROTOCOLLNPGM("start");
#endif //W25X20CL
	stdout = uartout;
	SERIAL_ECHO_START;
```

I'm not a coder but other threads suggested the "start" missing is a possible issue.

ogland, 2019-03-15:
I can't connect using pronterface on a PC either. When reverting to mk2s firmware it works flawlessly though. So it's something fishy going on in the mk2.5(s) firmware.

vaxxi, 2019-03-15:
MK3 here, I can confirm there's something wrong with the serial connection in 3.6 as well.

- 3.5.1, community build (@mionut with 7x7): OK

```
Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6aba8530, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=30.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo: 3.5.1-1778
Recv: echo: Last Updated: Jan 31 2019 10:33:09 | Author: (none, default config)
Recv: Compiled: Jan 31 2019
Recv: echo: Free Memory: 2002  PlannerBufferBytes: 1392
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect DISABLED
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
Recv: FSensor DISABLED
Recv:
Recv: echo:SD init fail
Recv: Error:No Line Number with checksum, Last Line: 0
Recv: Resend: 1
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
```
- official 3.6.0. fresh flash or after reboot: OK

```
Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x73bc0a10, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=30.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo: 3.6.0-2069
Recv: echo: Last Updated: Mar  1 2019 14:00:21 | Author: (none, default config)
Recv: Compiled: Mar  1 2019
Recv: echo: Free Memory: 2049  PlannerBufferBytes: 1392
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect DISABLED
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
Recv: FSensor DISABLED
Recv: 
Recv: echo:SD init fail
Recv: Error:No Line Number with checksum, Last Line: 0
Recv: Resend: 1
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
```

- latest code from MK3 branch, built today, connected right after firmware flash: OK

```
Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6ab9caf0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=30.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo: 3.6.0-2069
Recv: echo: Last Updated: Mar 15 2019 14:43:57 | Author: (none, default config)
Recv: Compiled: Mar 15 2019
Recv: echo: Free Memory: 2052  PlannerBufferBytes: 1392
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect DISABLED
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
Recv: FSensor DISABLED
Recv:
Recv: echo:SD init fail
Recv: Error:No Line Number with checksum, Last Line: 0
Recv: Resend: 1
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
```

- same self built firmware, turn printer off, turn it on again: NOK

```
Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x71e46950, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=30.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: echo: 3.6.0-2069
Recv: echo: Last Updated: Mar 15 2019 14:43:57 | Author: (none, default config)
Recv: Compiled: Mar 15 2019
Recv: echo: Free Memory: 2052  PlannerBufferBytes: 1392
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect DISABLED
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
Recv: FSensor DISABLED
Recv:
Recv: echo:SD init fail
Recv: echo:SD init fail
Recv: MMU not responding - DISABLED
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
```

The `Recv: start` is missing for the case where the connection is not working. @inquisitiveKangaroo I'd be happy to test further, just let me know what.

vaxxi, 2019-03-15:
Also, XYZ calibration seems to fix this problem.

Same firmware built today from MK3 branch. Turned printer on and off.

First connection attempt fails:

```
Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6ab91f70, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=30.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: echo: 3.6.0-2069
Recv: echo: Last Updated: Mar 15 2019 14:43:57 | Author: (none, default config)
Recv: Compiled: Mar 15 2019
Recv: echo: Free Memory: 2052  PlannerBufferBytes: 1392
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect DISABLED
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
Recv: FSensor DISABLED
Recv: 
Recv: echo:SD init fail
Recv: echo:SD init fail
Recv: MMU not responding - DISABLED
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
```

Only perform `Calibrate XYZ` from menu (no terminal log, as there is no serial connection) and the connection works. The `paused for user` is the prompt to perform first layer calibration.

```
Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6ab91dd0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=30.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo: 3.6.0-2069
Recv: echo: Last Updated: Mar 15 2019 14:43:57 | Author: (none, default config)
Recv: Compiled: Mar 15 2019
Recv: echo: Free Memory: 2052  PlannerBufferBytes: 1392
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect DISABLED
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
Recv: FSensor DISABLED
Recv: 
Recv: echo:SD init fail
Recv: echo:busy: paused for user
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: Error:No Line Number with checksum, Last Line: 0
Recv: Resend: 1
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Send: N1 M115*39
Recv: echo:busy: paused for user
Recv: echo:busy: paused for user
Recv: echo:busy: paused for user
Recv: echo:busy: paused for user
Recv: ok
Send: N2 M21*18
Recv: FIRMWARE_NAME:Prusa-Firmware 3.6.0 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Send: M105
Recv: Resend: 1
Recv: ok
Send: N1 M115*39
Recv: ok T:27.2 /0.0 B:27.4 /0.0 T0:27.2 /0.0 @:0 B@:0 P:29.4 A:33.1
Send: N2 M21*18
Recv: FIRMWARE_NAME:Prusa-Firmware 3.6.0 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
Send: M105
Recv: echo:SD init fail
Recv: ok
Recv: ok T:27.4 /0.0 B:27.6 /0.0 T0:27.4 /0.0 @:0 B@:0 P:29.4 A:33.0
Send: M105
Recv: ok T:26.9 /0.0 B:27.4 /0.0 T0:26.9 /0.0 @:0 B@:0 P:29.5 A:32.7
```

ogland, 2019-03-15:
Still the same issue with 3.7.0 RC1.

tonysmartantzoulatos, 2019-03-19:
@ogland you can't connect at all? Seems after the full reset I can connect and use all functions (simply I don't trust it at the moment) but I do have a large number of other issues.

ogland, 2019-03-25:
> @ogland you can't connect at all? Seems after the full reset I can connect and use all functions (simply I don't trust it at the moment) but I do have a large number of other issues.

I cant connect at all. Only for firmware flashing.

ogland, 2019-03-25:
I tried flashing with Slic3r PE now as someone suggested to me personally. Still the same issue.

Also I'm getting this during flashing:
`avrdude: stk500v2_command(): command failed`

tonysmartantzoulatos, 2019-03-25:
At the moment on the 2.5s I only had good success after resetting the board and running 3.7rc1. So far I can connect and run commands but since I got mk 2s to 2.5 I have stopped trusting printing with octopi

tonysmartantzoulatos, 2019-03-25:
Do you get success or fail at the end of fw update? I also got the avrdude fail on the beginning but at the end it said success

ogland, 2019-03-25:
Yeah this is super awkward but a factory reset was the magic bullet. Not the method for flashing or anything else.

closed 2019-03-25 by ogland