prusa3d/Prusa-Firmware #1631: Increase cooling time during unload.

Mjankor, 2019-03-14:
This is with a stock Mk3s extruder using single filaments. The MMU2 ramming and cooling moves will hopefully mitigate this problem for MMU prints.

I'm seeing a few reports (and the odd occurrence myself) of filament jamming during unload with the Mk3s extruder. What occurs is a blob forming just above the bondtech gears and jamming on the extruder body just above them.

This may be due to the shorter distance between hotend and gears. I suspect the plastic is still warm as it passes through the gears and is getting squished/bent there.

It may be worth increasing cooling time or cooling moves during unloads to ensure that the filament is cold when it gets to the gears and can't be bent out of shape.

PavelSindler, 2019-03-20:
Thank you for feedback. Does it jam immediately? You should actually remove filament after unload immediately before filament tip stiffens.

Mjankor, 2019-03-24:
Hi Pavel.

I'm going to close this bug, as I'm using it through an MMU, and I'm not entirely sure what the current firmware is up to. It may be solved already. The problem seemed to be where the MMU didn't pick up the slack immediately and (as you mentioned) remove the filament promptly during/after unload. 

It looks very similar to this bug, so I'll remove my report.
https://github.com/prusa3d/MM-control-01/issues/127

closed 2019-03-24 by Mjankor