prusa3d/Prusa-Firmware #1635: LCD MENU is very SLOW

birtalanelod, 2019-03-15:
LCD Menu is very slow, almost unusable with the rotary knob.
A lot of time happens I start the wrong print and have to wait for 0.5-1.5 s to move selection, or 4s to short files on the SD card
I loved my MK2 it was so fast, please make MK3 be similar.

F0x06, 2019-03-16:
Hi, i experienced same issue when i have a lot of files on SD card, try to have a reasonable amount of files inside the SD card, also make root folders to avoid sorting large amount of files.

But generally yes, the LCD can be very slow sometimes and it's really annoying.

TehH4rRy, 2019-03-16:
Agreed on the SD card clutter. I took the SD out and loaded a card with 5 files on. Printer speedy.

PavelSindler, 2019-03-20:
Yes, there is small pause before print (0.5 - 1.5s) which is caused by printer check if file is complete or not. We are reading last 20 kB of gcode file to check it and it takes something about 1 second to proceed. On MK2 it was little bit faster, because just 10kB was checked. However we had to enlarge it because gcode files generated by Slic3r changed over time.

PavelSindler, 2019-03-20:
To speeden up sorting files I would recommend having reasonable number of gcode files on SD card or if you want to have large number of files there, then use subfolders and keep reasonable amount of files in one folder is the best solution.  Also formating SD card from time to time is good idea. Sorting by time is faster on card which were formated recently.

closed 2019-04-09 by michalxfanta