prusa3d/Prusa-Firmware #1640: Z calibration during bed leveling

jerryhalstead, 2019-03-16:
If there's a bit of something under the bed or something else not quite right and Prusa detects that Z calibration is off during the bed leveling it errors out and after you press the button it just does a Z calibrate right where it failed. In some cases this might be right under the prusa filament holder and the extruder jams against it before reaching the top.

What about always homing first? If you're doing a normal xyz calibration that's what it does.

Crowlord, 2019-03-16:
2nd this. was hitting a leveling error and homing it manually fixed it.

DRracer, 2020-09-22:
@jerryhalstead If I understood right, his is the rare case when the machine does enforced Z-alignment exactly in the spot, where the filament holder sits at the top of the frame (the extruder hits it). 
So far there are no plans in changing the MBL algorithm (and/or the whole process) as there is more damage to cause...
Thank you for your understanding.

closed 2023-09-19 by github-actions[bot]