prusa3d/Prusa-Firmware #1641: LCD display overflows on large extruder values

sgoll, 2019-03-16:
When the extruder is at a position > ±999.9 and the user moves the extruder axis manually (using menu option "Settings"→"Move Axis"), the LCD display overflows and the current extruder position wraps into the fourth LCD display line (printing a lone digit at the beginning of that line and nothing else):

```

 Extruder:    +1234.

5
```

This can easily happen after a long print with the extruder running in relative mode.

The relevant function in the code seems to be `menu_draw_float31` in `menu.cpp` which does the formatting for numbers in format `+123.0` (`menu_draw_float13` probably suffers from the same problem in principle):

https://github.com/prusa3d/Prusa-Firmware/blob/cb1b070a58f994da42e2beb7ffe05365167df7af/Firmware/menu.cpp#L323

A possible solution would be to subtract any extra digits (necessary for printing the number) from the maximum text length, i.e. first get the formatted `length` of the number using format `%+06.1f`, then add up to `18 - length` characters from `text` in front of it.

All in all, this is a purely cosmetic problem, so probably low priority, but nonetheless it is still rather confusing at first for the user when he sees the wrapping from line 2 to 4.

PavelSindler, 2019-03-20:
Hi, thank you for reporting it.

niccoreyes, 2019-05-19:
I had problems as well with the live z adjustment in terms of text characters overflow. I only reverted the menu_draw_float13 and all references of the "13" back to the original format and functions to fix it on mine based on the history of changes. *sorry I'm still new on github and programming in general*

closed 2023-09-21 by github-actions[bot]