prusa3d/Prusa-Firmware #1642: 3.7.0 not issuing busy commands to octoprint

Crowlord, 2019-03-17:
On some functions busy commands are not issued to octoprint resulting in timeouts. 

This generally occurs when z calibration is triggered during bed mesh levelling (which happens A LOT) and this other issue has been raised separately by another user.

EXOgreen, 2019-03-18:
Can you post the number of that issue in here so they can be related? Thanks!

Crowlord, 2019-03-18:
its #1640

I believe a timeout may also occur due to 1646 (crash detected after mesh levelling)

dalgibbard, 2019-03-26:
Not 100% sure if related, but was printing the Linearity ECOR tower gcode, and octoprint timed out about 30% of the way in, causing a disconnect and failed print.

PavelSindler, 2019-03-29:
Our tests confirmed that busy messages are sent properly during Z-leveling on MK3, MK3S, MK2.5 and MK2.5S. Crash during mesh bed leveling should be fixed.

Crowlord, 2019-03-29:
Oddly even with repeated z Calibrations I have not re-encountered this issue. 

I did tweak timeout periods. 

With that in mind and for clarity to the dev team I believe this issue is closed.

closed 2019-03-29 by Crowlord