prusa3d/Prusa-Firmware #1645: Crash detected after 7x7 mesh bed level

mgg4, 2019-03-18:
### Environment
Printer: Prusa i3 Mk3 / MMU2
Firmware: 3.7.0-RC1 (Printer) / 1.0.5 (MMU2)

### Issue
After detecting the last point in a 7x7 mesh bed level operation, the printer moves to the home position.  The display shows "Crash detected", and the 7x7 mesh bed level operation repeats.  This second mesh bed level attempt succeeds and the print proceeds.

cyberboom2, 2019-03-18:
Mine do same its crash before starting 7x7 mesh points , after last point its go back into home position and  on screen say crash detected and restarting mesh leveling again, than printing.

Crowlord, 2019-03-18:
In addition this appears to cause me timeouts on octoprint (#1642) I have disabled it.

cosmith71, 2019-03-19:
I see this problem, too.  Not every time, but often enough that it is disabled.

PavelSindler, 2019-03-20:
Hi all. @cyberboom2 , @Crowlord , @cosmith71 on which printers did it happen to you? MK3 or MK3S?

cyberboom2, 2019-03-20:
Mk3

cyberboom2, 2019-03-20:
Mk3

Crowlord, 2019-03-20:
Mk3 makes 3!

thomasdlap, 2019-03-20:
Following along, switching to 7x7 results in the same for me on mk3s,
switched back to 3x3 and works fine.

cosmith71, 2019-03-20:
MK3.

leptun, 2019-03-20:
I also have a problem with a mk3. Don’t know if it’s related. It skips steps when going to the first point. It might be because of the fat bundle of cables at the back, but I’m not sure. Did not let it go through the mesh bed leveling. Reverted to 3.5.1 to avoid overheating board. Problem is gone, but there is only 3x3 mesh.

mgg4, 2019-03-25:
I had this issue every time I started a print.  Over the weekend, I had occasion to tear down my hot-end to clear a bit of broken filament.  After reassembly of the hot-end, I performed a full Wizard self-test and calibration.  Since that time, there have been no further crashes before the mesh bed leveling is started.

leptun, 2019-03-25:
I think the XYZ calibration is the one that fixed the issue

cyberboom2, 2019-03-25:
No XYZ calibration not fixed it i try to do that same problem crashing

michalxfanta, 2019-04-04:
Does this problem remain with the final FW 3.7.0?

Crowlord, 2019-04-04:
It seems fixed.

closed 2019-04-04 by michalxfanta

mgg4, 2019-04-04:
I haven’t experienced it, but that doesn’t mean the issue isn’t still there. You cannot prove a negative. 

--Mark

cyberboom2, 2019-04-04:
Its been release final 3.7.0 version its been fixed crashing problem

cyberboom2, 2019-04-04:
https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.7.0